mavlink/qgroundcontrol #5534: RSSI Percentage not working correctly on Pixhawk with SBUS port - ArduCopter

wallbanger123, 2017-08-07:
When using the SBUS port form RC RSSI, the indicator In the UI is reporting the value rather than converting to a percentage. Example: at full RSSI with my FrSky L9R it is reporting 255%. 

This is for ArduPilot only, it works correctly with PX4

OS X version tested.

flouwers, 2018-03-23:
I have the same issue: QGroundcontrol showing more than 250% RX RSSI.
In my case, the Receiver RSSI is sent through RC Channel 16.
Min & Max values are configured as 1500-2000 and I confirm CH16 values are within that range.

DonLakeFlyer, 2018-03-23:
What version of QGC and what version of firmware?

flouwers, 2018-03-23:
QGC 3.3.0 and Arducopter 3.5.5.
Same on Windows and Android

Kelly-Foster, 2018-06-21:
I am doing a thorough scrub on latest QGC / Windows 10 versus same ArduPlane firmware connected to MissionPlanner.   I am seeing RxRSSI = 0 on QGC, while same FC, same FW on MP shows 100%, which is what I expected.  I went to QGC's Parameters panel, and the RSSI params look good.  RxRRSI source is Dragonlink NanoRx, on ch12, range defined as 1000 to 1900, and ch12 is currently reading at 1900, so RxRSSI should be 100%, not zero.  Available to test when there is a fix.

dogmaphobic, 2018-06-22:
The error is that it is showing 255% when that value means "not supported". The spec calls for a value between 0-100 (%) and 255 for _invalid/unknown_

http://mavlink.org/messages/common#RC_CHANNELS_SCALED

dogmaphobic, 2018-06-22:
It's fixed in https://github.com/mavlink/qgroundcontrol/pull/6634 but not if you expect values greater than 100 to be meaningful. The original code was not properly handling _invalid/unknown_ values (255) and showing it as such.

>I went to QGC's Parameters panel

QGC takes the RX RSSI value from either the `RC_CHANNELS_SCALED` or the `RC_CHANNELS_RAW` messages. There are no parameters involved. If Ardupilot uses something else, I would need to know what and how it is encoded.

DonLakeFlyer, 2018-07-13:
Working as to spec now.

closed 2018-07-13 by DonLakeFlyer