mavlink/qgroundcontrol #5540: v3.2 and above not decoding UART packets on Windows Platform

LubinKerhuel, 2017-08-08:
Versions prior to 3.2 works great.
From v3.2 (tested 3.2.0, 3.2.2, daily build) with windows 7:
QgroundControl connect to UART. The FTDI cable is activated ; Setting a wrong baudrate provide the baudrate error.
But no packets are beeing shown on the Mavlink Inspector Widget.

Tested both V1 and V2 Mavlink packets using at minimum the HEARTBEAT packet. Reproductible on 2nd computer.

I tested UART to TCP bridge without success. 
Is it an issue related to QgroundControl build for Windows from 3.2.0 ?
Is there a minimum list of packet to activate the Mavlink Inspector ? (I had the param protocol as well as Way point protocol enabled)

Thanks,
Lubin

DonLakeFlyer, 2017-08-08:
Look here to see how QGC determines whether to create a vehicle or not: https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/MultiVehicleManager.cc#L75

closed 2017-08-08 by DonLakeFlyer

LubinKerhuel, 2017-08-08:
Hi Don,

You hit the nail on the head. I had HEARTBEAT set with Component Id = 0 (MAV_COMP_ID_ALL). QGC disregarded it. (this behavior was introduced in v3.2).
Component ID need to be 1 (MAV_COMP_ID_AUTOPILOT1) to make the vehicle registered within qgc, and its packets showing up within the mavlink inspector.
Many thanks for pointing-out the solution.

Mavlink Inspector not showing-up valid (CRC ok) packets is misleading as one tend to think there is a problem either with the packets/protocol implemented, or with qgroundcontrol library handling the low level communication… thus digging into a wrong direction which might be time consuming...

I read in another thread a suggestion to show-up all packets in the inspector, under a ‘disregarded packets’ name for example. That would help anyone with similar issue to dig right toward the correct solution.

Lubin