mavlink/qgroundcontrol #5577: RTSP streaming does not work when QGC has no internet access

SaliniVenate, 2017-08-16:
I'm trying to stream a video from drone to QGC running on ubuntu desktop over RTSP. The ubuntu machine is connected to drone over a wifi hosted by drone. Every time the ubuntu machine is disconnected from internet the QGC is not able to receive the video stream. However when I run vlc or gstreamer pipeline to receive the same video stream , it works. 
For video streaming should QGC be dependent on internet access?

DonLakeFlyer, 2017-08-16:
You must have something wrong in your setup. There is no depedancy on internet for video that I know of. 

@dogmaphobic Does this make any sense to you?

dogmaphobic, 2017-08-16:
No, it doesn't. The video connection is peer-to-peer with no dependency on anything else. 

I'm curious though, what is the URL used?

SaliniVenate, 2017-08-17:
The url used is : rtsp://192.168.8.1:8554/video13. This RTSP server is running in Intel Aero companion board.

1. With Internet : QGC is able to receive video streamed by drone
2. No Internet : QGC does not receive the video stream. Restarting the video (i.e choosing Video Stream Disabled and then choosing RTSP Video Stream in Settings) does not help. However VLC or gst pipeline is able to receive the video stream.
3. Internet disconnected while Video is streaming: QGC continues to receive video without disruption

The video manager and video reciever logs with internet disconnected is attached.
[logs_internet_disconnected.txt](https://github.com/mavlink/qgroundcontrol/files/1230194/logs_internet_disconnected.txt)

closed 2024-06-02 by HTRamsey