mavlink/qgroundcontrol #5593: libespeak.so.1 missing in AppImage

probonopd, 2017-08-19:
`QGroundControl: error while loading shared libraries: libespeak.so.1: cannot open shared object file: No such file or directory`

Solution:
Bundle libespeak inside the AppImage

Reference:
https://travis-ci.org/AppImage/AppImageHub/builds/266234023#L556

dogmaphobic, 2017-08-19:
There is a branch ready to merge that moves away from individual speech engines and uses Qt's own. It needs Qt 5.9 or greater so it's waiting until we update. Maybe it's time?

heschlie, 2018-01-20:
Is there an ETA on this landing, not able to launch this on Ubuntu 17.10.

DonLakeFlyer, 2018-01-20:
New daily builds are not working?

heschlie, 2018-01-21:
Ah missed the daily build link, those seem to work, thanks.

closed 2018-01-21 by DonLakeFlyer

probonopd, 2018-01-21:
I am referring to those on https://github.com/mavlink/qgroundcontrol/releases - any chance the fix will land there anytime soon? Thanks.