mavlink/qgroundcontrol #5598: Can't display video feed even though pixflow image is displayed

makquel, 2017-08-21:
I've been working on a communication pkg with several sensors for an custom aircraft. I'm using a TCP link to send telemetry info to the QGS, so I can visualise the camera feed on the pixflow tab, but can't see it on the video feed window

DonLakeFlyer, 2017-08-22:
Ok. Then I would assume something is wrong with you code. How is this a QGC bug?

makquel, 2017-08-22:
I have been experimenting with QGC, and I cannot seem to find any way to get live video to display without using mavlink. However, I can see the video streaming on the Pix4Flow tab, but can't display it on the video feed widget.
My camera driver is publishing a topic called /drone/image_raw_slow, and the following is the code that I used on my launch file in order to send it over mavlink to the QGC

  <node pkg="mavros_extras" type="gcs_image_bridge" name="gcs_image_bridge" clear_params="true" output="screen">
	<remap from="/mavlink/gcs_image/compressed" to="/drone/image_raw_slow/compressed" />
	<param name="image_transport" value="compressed"/>
	<param name="gcs_url" value="tcp://127.0.0.1:5760"/>
	<param name="gcs_url" value="tcp-l://127.0.0.1:5760"/>
  </node>

closed 2024-09-21 by HTRamsey