mavlink/qgroundcontrol #5602: qrc:qml/FirmwareUpgrade.qml:115: TypeError: Type error

KhristopherK, 2017-08-23:
Firmware cant be uploaded to pixhawk. Setup:
- Pixracer
- QgroundControl HEAD: 08deff4b7 2017-08-19

Error:
qml: 0.03518518518518519
qml: 34.5 38
Orphaned log file count 0
QDBusConnection: name 'org.freedesktop.Geoclue.Master' had owner '' but we thought it was ':1.93'
qrc:qml/FirmwareUpgrade.qml:115: TypeError: Type error

Tried PX4 stack, and all APM firmwares. Same result

DonLakeFlyer, 2017-08-23:
Can you try a prebuilt AppImage version?

xoltar, 2017-08-24:
I had this problem (along with other firmware upgrade failures that did not result in a message on the console) on **Ubuntu 17.04** with yesterday's daily appimage build. The same day's build worked fine for upgrading firmware on **Windows**. This was with pixhawk mini.

DonLakeFlyer, 2017-08-24:
Could you check to see if this is daily build only as opposed to Stable as well?

KhristopherK, 2017-09-10:
Hi @DonLakeFlyer I have tried now on the latest daily build, as in 2017 09 09 and having still the same problem. Can't upload firmware. I have now tried on OSX and on Ubuntu 16.04 with the same result.I am using prebuilt appimage versions. I am also having an issue where, even if I manually upload the firmware using command (via make px4 upload) I have issues connecting to the gcs via usb. This is with a pixracer (but get the same issue with a Pixhawk, and a pixfalcon)

DonLakeFlyer, 2017-09-14:
@KhristopherK What about Stable build?

sritank, 2020-01-21:
I'm having the same issue using pixracer with QGroundControl 3.5.6 on Ubuntu 18.04. I am able to successfully upgrade the firmware on the board and it reboots. However, QGC fails to reconnect so I don't get the setup options for Airframe, Radio, etc. 
Didn't face this issue on Windows 10