mavlink/qgroundcontrol #5609: QGroundControl Crash in "Change command"

flo-vienna, 2017-08-27:
Used Version: QGroundControl Version 3.4.2 
on device Vido M82 Pro  with Android vers. 5.1 (Kernel 3.10.65+ , MOLY.LR9.W1444.MD.LWTG.MP.V12.P2 Baseband)  

Way to reproduce:
Starting the app, (no vehicle connection, no video capture card connected, no stream available etc. ), camera access / location access granted.
Zooming to some area (place does not mater)
Creating a survey grid. (Camera Canon Powershot S100) with 70% overlapp

Selecting the newly created Survey command -> Change Command -> Crash

Attached the output from Catlog:

```
08-28 14:30:13.367 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemWidth"
08-28 14:30:13.367 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemWidth"
08-28 14:30:13.367 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemHeight"
08-28 14:30:13.368 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemHeight"
08-28 14:30:13.372 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemWidth"
08-28 14:30:13.372 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemWidth"
08-28 14:30:13.373 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemHeight"
08-28 14:30:13.373 W/QGroundControl(9753): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117 ((null)): qrc:///qml/QGroundControl/Controls/MissionItemIndexLabel.qml:117:5: QML QGCMouseArea: Binding loop detected for property "_itemHeight"
```

Log entries that appear to be related to the crash:

```
08-28 14:31:14.758 F/libc    (9753): Fatal signal 11 (SIGSEGV), code 1, fault addr 0x44 in tid 9781 (QtThread)
08-28 14:31:14.881 W/CameraService(246): Disconnecting camera client 0xf3c368f0 since the binder for it died (this pid 246)
08-28 14:31:14.882 W/MtkCam/CamClient/FDClient(246): (246)[stopFaceDetection] FD was not running
08-28 14:31:14.882 W/MtkCam/CamClient/OTClient(246): (246)[stopOT] OT was not running
08-28 14:31:14.895 W/InputDispatcher(786): channel '28a1c7a6 org.mavlink.qgroundcontrol/org.mavlink.qgroundcontrol.QGCActivity (server)' ~ Consumer closed input channel or an error occurred.  events=0x9
08-28 14:31:14.895 E/InputDispatcher(786): channel '28a1c7a6 org.mavlink.qgroundcontrol/org.mavlink.qgroundcontrol.QGCActivity (server)' ~ Channel is unrecoverably broken and will be disposed!
08-28 14:31:14.903 W/ActivityManager(786): Force removing ActivityRecord{1d1ca40 u0 org.mavlink.qgroundcontrol/.QGCActivity t7}: app died, no saved state
08-28 14:31:14.907 W/WindowManager(786): Force-removing child win Window{2c0d1c90 u0 PopupWindow:2a32bb66 EXITING} from container Window{28a1c7a6 u0 org.mavlink.qgroundcontrol/org.mavlink.qgroundcontrol.QGCActivity}
08-28 14:31:14.906 W/InputDispatcher(786): Attempted to unregister already unregistered input channel '28a1c7a6 org.mavlink.qgroundcontrol/org.mavlink.qgroundcontrol.QGCActivity (server)'
08-28 14:31:14.926 W/ResMgrDrv(246): [CloseHdmi]WRN(  378):HDMITX kernel open fail, errno(2):No such file or directory
08-28 14:31:14.926 W/MtkCam/devicemgr(246): [getBootMode] fail to open: /sys/class/BOOT/BOOT/boot/boot_mode
08-28 14:31:14.927 W/MtkCam/ResManager(246): [close] No user
08-28 14:31:14.937 W/System.err(786): java.lang.NullPointerException: Attempt to invoke interface method 'android.os.IBinder com.android.internal.app.IAppOpsCallback.asBinder()' on a null object reference
08-28 14:31:14.937 W/System.err(786): 	at com.android.server.AppOpsService.stopWatchingMode(AppOpsService.java:540)
08-28 14:31:14.937 W/System.err(786): 	at com.android.internal.app.IAppOpsService$Stub.onTransact(IAppOpsService.java:122)
08-28 14:31:14.937 W/System.err(786): 	at android.os.Binder.execTransact(Binder.java:451)
08-28 14:31:14.991 W/InputMethodManagerService(786): Got RemoteException sending setActive(false) notification to pid 9753 uid 10094
08-28 14:31:14.995 E/GED     (1431): Failed to get GED Log Buf, err(0)
```

Greets

flo-vienna, 2017-10-05:
Anything missing or why is nobody caring about that ?

DonLakeFlyer, 2018-01-04:
Fixed a while ago

closed 2018-01-04 by DonLakeFlyer