mavlink/qgroundcontrol #5610: Problem with log download from Pixhawk (Reproducible)

samacumen, 2017-08-28:
**Summary:** Unable to download logs via QGC connected to Pixhawk beyond 5th attempt (over USB)
**Firmware:** Stable v3.2.2 QGC, v1.6.5 Stable Release PX4, Ubuntu 16.04LTS 

**Steps to reproduce:**
1) Open the "Log Download" tab in QGC
2) Press 'Refresh' button - brings the list of logs
3) Download any log (choose a small size so that its faster to reproduce the issue)
4) Wait until 'Downloaded' state
5) Repeat steps 2), 3) and 4)  judiciously,at least 5 times.

On the 5th trial, on pressing the 'Refresh' button "no logs are obtained" from PX4.

**How it must be on pressing "Refresh":**

[image]

**On 5th attempt (No Logs obtained):**

[image]

This issue very much reproducible at my end when I have attempted the above steps as is. I am not sure if this is a Pixhawk or a QGC issue. My suspect is that the state in PX4 goes into a wrong state during these operations. Since this is observed at QGC side, I am reporting the issue here.

Request your observation: @LorenzMeier , @dogmaphobic

samacumen, 2017-08-28:
**Another issue I found with log download is:**

If the "Status" of a log goes into "error" (because a log file was corrupt in the Pixhawk's SD card), Pixhawk needs to be rebooted to show up the list of logs again for download. If an error occured for a log download, I would expect to be able to request the list of logs for download, anyways.

**See the first line below, press "Refresh" after error, nothing shows up again. Requires Pixhawk reboot:**

[image]

samacumen, 2017-09-22:
Is this being addressed and any idea if it could be resolved for the next release? Thanks.

DonLakeFlyer, 2017-09-25:
@bkueng Is this different from what you fixed?

bkueng, 2017-09-26:
Yes, these look like different issues

DonLakeFlyer, 2017-09-26:
@bkueng Could you take a look at this one then as well? Seems a bit similar to changes you already made. I assume this is QGC side as opposed to Firmware side. but maybe not?

bkueng, 2017-09-28:
Ok, I'll try to reproduce

bkueng, 2017-10-02:
First issue: I tried more than 10 times on QGC master, PX4 master on a Pixracer via USB, but never had an issue.

Second issue: should be fixed in https://github.com/mavlink/qgroundcontrol/pull/5722

samacumen, 2017-10-02:
@bkueng First issue is probably more concerning. I just tried it again this morning and it is very much reproducible at my end (as given from Steps (1) to (5) ), following it as is. After downloading a log, you need to click 'Refresh' and it should reproduce on 6th cyclic attempt, i.e. receive nothing from the Pixhawk (running PX4 master).

Let me know if you need more information in order to reproduce this.

Thanks.

AlexisTM, 2017-10-06:
I had to delete all logs (465) and refly and it downloaded the log.

QGC was spamming: Received log data for wrong log

bellahduan, 2017-10-10:
@samacumen hi , the log download from QGC is **.px4log , but it can not use to analysis.  how to solve the problem?

DonLakeFlyer, 2017-10-20:
@bkueng Ok, to close this issue now?

bkueng, 2017-10-21:
No, according to @samacumen there's one more issue (which I'm unable to reproduce):

> Unable to download logs via QGC connected to Pixhawk beyond 5th attempt (over USB)
>
> Steps to reproduce:
>
>   - Open the "Log Download" tab in QGC
>   - Press 'Refresh' button - brings the list of logs
>   - Download any log (choose a small size so that its faster to reproduce the issue)
>   - Wait until 'Downloaded' state
>   - Repeat steps 2), 3) and 4) judiciously,at least 5 times.
>
> On the 5th trial, on pressing the 'Refresh' button "no logs are obtained" from PX4.

fkaiser, 2017-10-23:
Observed similar issue as described by @samacumen. The problem seems to lie on the PX4 side with this message showing up as the debug output from autopilot:

[mavlink] MavlinkLogHandler:: init Error creating /fs/microsd/$log$.txt

Message stems from this line in [PX4 code](https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_log_handler.cpp#L467).

In that case there seems to an issue with the file handling. Only remedy is to reboot autopilot.

samacumen, 2017-10-23:
@fkaiser Thanks for helping reproduce the issue. You probably hit the birds eye! I was also quite sure in my analysis of the issue being on PX4 side as I found the QGC side clean since it only acts on the received mavlink messages from PX4. At certain (repeated) point of action, QGC stops receiving.

Should we transfer this to the PX4 team? @LorenzMeier

DonLakeFlyer, 2017-10-23:
> Should we transfer this to the PX4 team?

@bkueng Any comments on this?

bkueng, 2017-10-23:
Hmm, the error handling on the Firmware looks correct to me. It seems to be an underlying issue in the file system or NuttX or not enough file descriptors. And QGC then reacts correctly showing an empty list. So yes we can move it over, but I'm not sure if it helps.

@samacumen @fkaiser if you're in a position to recompile, can you exchange https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_log_handler.cpp#L467 with:
```
PX4LOG_WARN("MavlinkLogHandler::init Error creating %i %s\n", errno, kTmpData);
```
And see what the output is?
If it's a file descriptor limit, it's resolved on PX4 master - which seems to me the most likely cause.

bkueng, 2017-10-23:
Just came across another bug: https://github.com/mavlink/qgroundcontrol/pull/5778

samacumen, 2017-10-23:
@bkueng Thanks for your analysis and taking notice of these issues. As a user of the system, I could reproduce it and don't mind if qgc or px4 side fixes it as appropriate.

I'll be back to work in a couple days and will compile what you suggest and report back with the result.

samacumen, 2017-10-27:
I am unable to reproduce the issue reported with the steps I initially wrote, not sure if it has been fixed in the interim. I used the latest PX4 master and latest qgc_master branches as of today.

@fkaiser you said you managed to reproduce this issue, which branch did you reproduce this problem on?

fkaiser, 2017-10-30:
I think it was stable v1.5.5 on the PX4 side.

DonLakeFlyer, 2017-12-30:
The combination of QGC and Firmware fixes should have taken care of this

closed 2017-12-30 by DonLakeFlyer