mavlink/qgroundcontrol #5613: Problems with survey

juvinski, 2017-08-28:
Hi,

drawing an irregular area on Survey is causing erroneous flight lines.
on the image irregular_90_degrees if you change from 0 to 90 degrees area of interest, the lanes are not enough to complete the survey.
On the lanes_greather_than_necessary image the lanes are larger than necessary - the lines are far from the area of interest.

[image]
[image]

Regards

DonLakeFlyer, 2017-08-29:
Can you attach the .plan file for the first survey here and I'll take a look. Just zip it up and attach. Something is going wrong there.

The second survey is a concave polygon which QGC does not support. It will turn it into a convex polygon and create a survey around that.

juvinski, 2017-08-29:
Hi @DonLakeFlyer ,

for sure, I'm attaching the 2 files:
Can I help you ?
[mission_90_degrees.plan.zip](https://github.com/mavlink/qgroundcontrol/files/1259013/mission_90_degrees.plan.zip)

zorandragan, 2017-11-09:
Hi @DonLakeFlyer 

> The second survey is a concave polygon which QGC does not support.

Why is this the case with QGC?

DonLakeFlyer, 2017-11-09:
Fixed:
[image]

> Why is this the case with QGC?

It is not supported because the code for survey generation is not currently capable of generating correct surveys for concave polygons.

DonLakeFlyer, 2017-11-09:
I added a separate issue for convex polygons: https://github.com/mavlink/qgroundcontrol/issues/5830

closed 2017-11-09 by DonLakeFlyer