mavlink/qgroundcontrol #5616: battery amps per volt

Stifael, 2017-08-30:
I did not understand how I should fill out this field (the calculation gives me Inf). 
In addition I was not able to read the instruction, since on desktop it is very tiny.

ndepal, 2018-01-16:
I also do not understand how I should fill out the "Amps per Volt" field in the QGC "Power" tab.

In order to compute a reasonable value for the current draw scaling, I assume I have to have a decent amount of current (this is also recommended in the APM docs). For this I wanted to arm the vehicle and increase the thrust until I saw some amps flowing. However, when the vehicle is armed I am locked out of the QGC settings, so I can't set this.

I was trying to get https://github.com/PX4/Firmware/pull/8153 to work properly. @MaEtUgR does this need to be set for your PR? If so, how are you setting this?

DonLakeFlyer, 2018-01-16:
You use the calculator while the vehicle is disarmed.

@Stifael Is this still not working for you in latest daily builds?

ndepal, 2018-01-16:
The problem is that when it's disarmed, there is no current to speak of. Both the value shown in QGC and the one shown on the external amp meter are very close to zero. This won't give a good calibration.

I am basing this off of the following found in the APM docs. It describes an inherent property of power modules and should therefore also apply to PX4/QGC.

>Most current sensors are not very accurate at low currents (less than 3Amps). Typically you should perform current calibration at around 10A. The exception is PMs that use hall-effect sensors, like those from Mauch.

DonLakeFlyer, 2018-01-16:
You'd have to ask the PX4 firmware folks about that.

MaEtUgR, 2018-01-16:
@ndepal It does not need to be set, I think there are even power modules with no current reading. And it should also be able to give you a rough battery estimate without current reading. But to set the parameters correctly and use the current reading is recommended because you can get much more meaningful and reliable battery estimation. I'm preparing docs for battery settings https://github.com/PX4/px4_user_guide/pull/131

Honestly I never set the `BAT_A_PER_V` but I would:
- Detach propellers
- setup the vehicle with a power circuit through an ampere meter
- arm and note the steady state current
- set the parameter such that the current reported by PX4 is the same as the measured one
(at lest for the voltage devider there's logic in QGC that does the calculation for you).

ndepal, 2018-01-16:
Thanks @MaEtUgR 

I'm interested in setting everything right to get the best estimate possible.

> set the parameter such that the current reported by PX4 is the same as the measured one
(at lest for the voltage devider there's logic in QGC that does the calculation for you).

That exists also for the amp per volt parameter, but, just like for the voltage divider, it asks you to enter the amps currently measured externally and compares that to the currently measured current as reported by the flight controller. From those two values it then re-sets the parameter value (at least that's what I think happens). This works fine for the voltage divider, but for the amp parameter the problem is that you can only get a meaningful external measurement when armed and throttled up, but you can only change the parameter when disarmed (i.e. when the internal current measurement to which your supplied value will be compared is 0).

A workaround would be (maybe that's what you're suggesting) to iteratively
- arm
- note both internal and external amp readings
- disarm
- increase/decrease the parameter
- repeat 

until the two readings match.

I believe disabling the "lock out of settings when armed" feature in QGC for the Power tab would fix the issue.

MaEtUgR, 2018-03-26:
@dogmaphobic This seems to be traced back to a UI issue:
The current measurement calibration can not be accessed while armed but only when armed there's a useful current to calibrate the measurement.

Any update on this? If the workflow was fixed already then please close or else assign/tag.

DonLakeFlyer, 2018-03-26:
> The current measurement calibration can not be accessed while armed but only when armed there's a useful current to calibrate the measurement.

Not according to how this previously worked. Did something change here? @LorenzMeier Something change in batt calibration?

DonLakeFlyer, 2018-03-26:
That said, this can change to not not allow while flying instead of armed I think. But still get to get clarification on how you should use this with PX4.

closed 2018-03-27 by DonLakeFlyer