mavlink/qgroundcontrol #5621: A-UAV x2.1: No Voltage, nor Amp reading in the QGC.

tubeme, 2017-08-31:
With regular FMUv2 no problems everything is fine. If A-UAV x2.1 the voltage and amp reading stay at -1.

Not sure if it is PX4 or QGC problem?

DonLakeFlyer, 2017-09-01:
Those values come from the firmware.

closed 2017-09-01 by DonLakeFlyer

tubeme, 2017-09-25:
It turns out that it is QGC problem.

Related to this: https://github.com/PX4/Firmware/issues/7931#issuecomment-328241891

I was not able to change the value for the Voltage devider in the Power menu in the Settings. Nor I could calculate one. It did not accepted the value. By default the voltage divider stays at 0.00000 as is the bat cell count and the apm divider.

So I had to forcefully change the BAT_V_DIV to some value different than 0.0000. And then everything started working. I could calculate the new voltage divider.

This values were never preset to this controller after flashing the 1.6.5 and resetting the frame. 

So it was like brand new installation and not upgrade of px4. So all the values were 0.00 in the bat calibration.