mavlink/qgroundcontrol #5632: Please Help/ How to Build on Windows

mrAyaz, 2017-09-04:
How to configure Compile?

chodjo, 2017-09-04:
Hello,

Please follow [https://github.com/mavlink/qgroundcontrol/blob/master/README.md](instructions)

mrAyaz, 2017-09-04:
i have this error.

jom: C:\Users\Mamedli Ayaz\Documents\GCS\build-qgroundcontrol-Desktop_Qt_5_7_1_MSVC2015_32bit-Release\Makefile.Release [moc\moc_qextserialenumerator.cpp] Error 1
gstqtvideosinkmarshal.c
jom: C:\Users\Mamedli Ayaz\Documents\GCS\build-qgroundcontrol-Desktop_Qt_5_7_1_MSVC2015_32bit-Release\Makefile [release] Error 2
19:54:54: The process "C:\Qt\Qt5.7.1\Tools\QtCreator\bin\jom.exe" exited with code 2.
Error while building/deploying project qgroundcontrol (kit: Desktop Qt 5.7.1 MSVC2015 32bit)
The kit Desktop Qt 5.7.1 MSVC2015 32bit has configuration issues which might be the root cause for this problem.
When executing step "Сборка"

mrAyaz, 2017-09-04:
Please help how to fix?

closed 2017-09-04 by DonLakeFlyer