mavlink/qgroundcontrol #5634: FlightDisplayView not showing PlanElemementMasterController

chodjo, 2017-09-05:
When selecting an APM ArduPlane vehicle mock link, FlightDisplayView doesnt show PlanElemementMasterController menu.

DonLakeFlyer, 2017-09-08:
Not sure what part of the UI your are talking about?

chodjo, 2017-09-12:
The FLY menu, with Takeoff, RTL, Pause and Action options is not displayed when using an APM ArduPlane mock link.

DonLakeFlyer, 2017-09-13:
Do you mean disabled? Or are they missing from the screen? ArduPlane guided commands are not fully supported yet. So most will be disabled.

chodjo, 2017-09-13:
Missing from screen.

DonLakeFlyer, 2017-09-13:
What OS and what version of QGC. It's all working fine for me.

chodjo, 2017-09-13:
OS Linux, and development version downloaded 2017-08-01. Maybe my fault then, as Im trying to understand the code and may have changed something then. Sorry for the inconvenience.

DonLakeFlyer, 2017-09-13:
If you can reproduce this with a pre-buit AppImage version let me know,.

closed 2017-09-13 by DonLakeFlyer