mavlink/qgroundcontrol #5646: Add joystick buttons support

kikislater, 2017-09-12:
Hi, 
Since this PR https://github.com/ArduPilot/ardupilot/pull/6927 Qgroundcontrol could handle joystick. 
It's working well but buttons are not supported. Is it possible to add button support for Ardupilot firmware with joystick. Idea is to arm/disarm, change flight mode with joystick buttons.
I have only this message : 
Buttons 0-15 reserved for firmware use

[image]

jaxxzer, 2017-09-15:
This needs to be implemented in the autopilot firmware first. Which autopilot firmware (and version) are you using?

kikislater, 2017-09-16:
Ok didn't think it was a fw issue ! 
I'm running ardupilot fw :  arduplane 3.8.2

jaxxzer, 2017-09-20:
https://github.com/mavlink/qgroundcontrol/issues/5644

gigorvd, 2018-01-09:
I tested the firmware on ArduPlane and ArduCopter and it does not work => Buttons are not supported. But on ArduSub working correctly.

JAR4x4, 2018-03-13:
Any update on this matter?

DonLakeFlyer, 2018-03-13:
@jaxxzer Any update on what the status of this is supposed to be for other ArduPilot firmware types?

JAR4x4, 2018-03-13:
I've got the four control axis working, but I'm not able to map any buttons. I'm running ArduCopter 3.5 on a Navio2

jaxxzer, 2018-03-13:
All of the ardupilot vehicles support MANUAL_CONTROL for r,p,y,throttle input. Only ArduSub supports the use of buttons for other inputs.

JAR4x4, 2018-03-13:
@jaxxzer thanks for the reply! Any chance to get buttons to work with ArduCopter?

jaxxzer, 2018-03-13:
No issue in QGC. @JAR4x4 this would need to be added as a feature request in the ardupilot repository. I don't think this will happen any time soon as arducopter is built around the concept of RC transmitter switches which have fundamentally different operation (latching) than momentary buttons.

DonLakeFlyer, 2018-03-13:
Closing this then

closed 2018-03-13 by DonLakeFlyer