mavlink/qgroundcontrol #5650: Qgroundcontrol (macOS, 3.2.4/daily) fails to display FPV video.

kenkeiter, 2017-09-13:
When using qgroundcontrol (macOS `10.12.4`, Qgroundcontrol versions `3.2.4` and `daily`) in conjunction with the instructions in the [video streaming guide](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md), display of video fails due to an "internal data flow error."

I've tried the example `gstreamer` pipelines in the documentation (and verified that I'm seeing streamed data using `nc`). I've also tried variations with different h264 encoders to ensure that I'm not just seeing issues with the way a specific encoder has been built for macOS.

I installed gstreamer and associated plugins using the following:

```sh
$ brew install gst gst-plugins-good gst-plugins-bad
$ brew install gst-plugins-ugly --with-x264
```

Please let me know if I can be of service in debugging. The log output was as follows (for both versions):

```
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/comm/MAVLinkProtocol.cc:413 - "Orphaned log file count 0"
[!] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:498 - "Internal data flow error."
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:451 - "Gstreamer error!"
[!] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:498 - "Internal data flow error."
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:451 - "Gstreamer error!"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:423 - "No pipeline"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:206 - "start()"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:478 - "State changed, _streaming: false"
[D] at /Users/travis/build/mavlink/qgroundcontrol/src/VideoStreaming/VideoReceiver.cc:379 - "Running"
```
(the aforementioned log excerpt repeats)

kenkeiter, 2017-09-13:
Also wanted to note that I was able to validate my encoding setup a little further. By streaming with the following command:

```
gst-launch-1.0 videotestsrc pattern=ball ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600
```

I am able to watch the ball video in VLC player by using an `.sdp` file with the following contents:

```
v=0
m=video 5600 RTP/AVP 96
c=IN IP4 127.0.0.1
a=rtpmap:96 H264/90000
```

DonLakeFlyer, 2017-09-13:
You're better off with this stuff on Gitter where there are more video people. Since it is most likely a setup problem as opposed to a QGC bug.

kenkeiter, 2017-09-13:
@DonLakeFlyer If you feel that, given the information above, it's still a setup issue, I'll certainly give that a shot. Thank you.

If you don't mind, I'd like to leave the ticket open for just a while until I verify w/the Gitter folks.

lbegani, 2017-09-14:
The error is perhaps due to unsupported format by QGC. QGC doesnt support all the formats supported by gstreamer but limited few

kenkeiter, 2017-09-14:
I wanted to note that @lbegani resolved my issue, which was caused by incorrect documentation. The test command listed in the [video streaming doc](https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md) suggests using the following command to test locally: 

```
gst-launch-1.0 videotestsrc pattern=ball ! x264enc ! rtph264pay ! udpsink host=127.0.0.1 port=5600
```

This is incorrect. The following command may be used, instead.

```
gst-launch-1.0 videotestsrc ! video/x-raw,width=640,height=480 ! videoconvert ! x264enc ! rtph264pay ! udpsink host=0.0.0.0 port=5600
```

When I have a second, I'll submit a PR to resolve the issue.

Thanks again for your help, @lbegani! 🎊 💯

jaxxzer, 2017-09-14:
The command in the documentation is correct. If 0.0.0.0 works, but 127.0.0.1 does not, I suspect something is up with your network configuration. I'm not sure what would cause this, what's your OS?

kenkeiter, 2017-09-14:
@jaxxzer Nothing's up with my network config. The unicast address is not the only thing in the aforementioned doc that changed -- an additional item was added to the pipeline (`video/x-raw,width=640,height=480`). Changing the unicast address back to `127.0.0.1` works as well, given that QGroundControl binds to `0.0.0.0` (all IP addresses on the local machine).

I have updated my suggested doc change to revert back to using `127.0.0.1` as the host address. My apologies; that part of the change was unnecessary.

jaxxzer, 2017-09-14:
Hm, the original command works fine for me on Ubuntu.

kenkeiter, 2017-09-15:
@jaxxzer Yeah, no idea why it wouldn't on macOS. Software, amirite?

closed 2026-10-04 by github-actions[bot]