mavlink/qgroundcontrol #5658: Qgroundcontrol won't run after installed

raygamero, 2017-09-15:
On two computers running windows 7 and windows 10, I downloaded multiple installers. https://docs.qgroundcontrol.com/en/releases/daily_builds.html
and from https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html
All attempts failed with the message This application failed to start because it could not find the qt platform in the plugin "windows" in "". 

I uninstalled, reinstalled, and restarted.

raygamero, 2017-09-15:
This appears to be localized to these two particular computers.  Common factors are virtual com ports and mission planner.  Installed on another computer and no problems.

DonLakeFlyer, 2017-09-15:
Hmm, strange stuff. @NaterGator Ever seen anything like this?

DonLakeFlyer, 2017-09-15:
May be OpenGL drivers, but doesn't feel like it. Have you tried starting QGC on these machines use the two GPU safe mode links?

NaterGator, 2017-09-15:
Sorry been a little bit out of the loop with Hurricane Irma in the area. This sounds like an installation problem on machines with either custom defined qt environment variables or the qt SDK. This document describes the typical search order of qt plugins: http://doc.qt.io/qt-5/windows-deployment.html#qt-plugins

DonLakeFlyer, 2017-09-15:
Ooh, right forgot about that Nate. Hope you are doing well.

NaterGator, 2017-09-15:
All good now, thanks! @raygamero can you comment on if these machines have another version of Qt installed at C:\Qt or if they are used for development purposes with Qt apps?

raygamero, 2017-09-17:
Thanks, Nate.  I'll try this on Monday or so.  Not a hug issue right now but will let you know how we resolve it.

HanJIk, 2017-09-18:
help me!!
(QGroundControl.exe:5768): GLib-ERROR **:gmem.c:103: failed to allocate 3110491 bytes
Microsoft Visual C++ Runtime Library
 Runtime Error!
Program: C:\Program Files (x86)\qgroundcontrol\QGroundControl.exe

Specifications
 win8.1 64bit
firmware ardusub3.5.1
 QGC version v3.2.4

please

DonLakeFlyer, 2017-12-29:
@raygamero Any update on this?

closed 2018-01-03 by DonLakeFlyer