mavlink/qgroundcontrol #5664: Crashing the waypoints of multi vehicles

gizemsungu, 2017-09-18:
I have 2 vehicles connected by different TCP ports and I load different waypoints to each vehicle. When each vehicle starts their missions in "Auto" mode, two of them refer the waypoints of the second vehicle. 
For example:
Vehicle 1 has the waypoints at left side of the image and Vehicle 2 has the waypoints at the right side. 
[image]

When these two vehicles start their missions, two of them go to the second waypoint of Vehicle 2. 
[image]

Is there any solution for this problem?
Thanks

DonLakeFlyer, 2017-09-18:
What SITL simulator are you using?

gizemsungu, 2017-09-18:
ardupilot SITL simulator http://ardupilot.org/dev/docs/sitl-simulator-software-in-the-loop.html

DonLakeFlyer, 2017-09-18:
I've had problems before with ArduPilot SITL multi-instance before where messages get routed to the wrong place. That sounds like the same problem you may be having not sure. Also keep in mind that when Planning mission the Plan view is interacting with the currently active vehicle (selected in toolbar). It can get confusing and if you switch active vehicles you have to manually sync the Plan view with the newly selecteed vehicle.

I'll try some steps to reproduce myself. But I'm not sure this is a QGC side problem yet.

gizemsungu, 2017-09-20:
Which SITL simulator do you suggest to use with QGC?

DonLakeFlyer, 2017-09-20:
There isn't really a choice. If you are using ArduPilot you need to use their simulator.

DonLakeFlyer, 2017-12-30:
This the the MAVProxy multi-vehicle problem. If you run ArduPilot SITL without MAVProxy it will work fine.

closed 2017-12-30 by DonLakeFlyer