mavlink/qgroundcontrol #5670: QgroundControl Running Error

HanJIk, 2017-09-19:
(QGroundControl.exe:6776): GLib-ERROR **:gmem.c:103: failed to allocate 3110491 bytes

Microsoft Visual C++ Runtime Library
 Runtime Error!
Program: C:\Program Files
 (x86)\qgroundcontrol\QGroundControl.exe

This application has requested the Runtime to Terminate it in an unusual way.
Please contact the application's support team for more infomation.

my computer Specifications
 win8.1 64bit
 firmware ardusub3.5.1
 QGC version v3.2.4

please

HanJIk, 2017-09-19:
And QgroundControl (GPU Compatibility Mode) What is the difference between QgroundControl (GPU Safe Mode)?

NaterGator, 2017-09-19:
Hi @HanJIk. GPU compatibility mode uses ANGLE to render the UI for devices with incompatible OpenGL drivers. The Qt information for this mode is here: http://doc.qt.io/qt-5/windows-requirements.html#dynamically-loading-graphics-drivers
For the most part we detect this and apply the most appropriate solution automatically at startup. 

This error is coming from glib, and as far as I know the only glib component we use on Windows is gstreamer. Do you have gstreamer installed elsewhere on your system for another program?

HanJIk, 2017-09-19:
[image]
gstreamer Related files. When you look at this history
I do not think it's installed yet.

HanJIk, 2017-09-19:
How do I install gstreamer?

NaterGator, 2017-09-19:
Can you try removing the file registry.i686.bin located at %USERPROFILE%\AppData\Local\Microsoft\Windows\INetCache\gstreamer-1.0?

DonLakeFlyer, 2017-09-19:
@NaterGator Is there some way to force Windows/Qt to only use the libraries we provide? Both in this case and the Qt installed somewhere else case?

NaterGator, 2017-09-19:
In this case I believe we can use a combination of [GST_REGISTRY](https://cgit.freedesktop.org/gstreamer/gstreamer/tree/gst/gstregistry.c?id=1.5.2#n1715) and [GST_PLUGIN_PATH](https://cgit.freedesktop.org/gstreamer/gstreamer/tree/gst/gstregistry.c?id=1.5.2#n1595) environment variables to control plugin loading behavior. 

In almost every normal circumstance the Qt plugin case is handled correctly because of windeployqt. We could use [qt.conf](http://doc.qt.io/qt-5/qt-conf.html) embedded in a resources file to specify our own immutable plugin search paths if we suspect that user problems are more widespread.

jaxxzer, 2017-10-09:
Same thing happening here: https://github.com/mavlink/qgroundcontrol/issues/5311

jaxxzer, 2017-10-09:
@NaterGator can you help me with setting up windows stack trace/log dumps for debugging this?

jaxxzer, 2017-10-20:
@HanJIk is this still happening for you? You can follow the instructions [here](http://qgroundcontrol.org/bugreporting#reporting_crashes_from_windows_builds) to get a crash dump that we can look at.

closed 2024-06-02 by HTRamsey