mavlink/qgroundcontrol #5672: Full Parameter Load fails

tops4u, 2017-09-19:
I was trying Arducopter and saved all Parameters from 1.6.2dev in a file. After Re-Installing PX4 1.6.5dev (Master) I tried to reload all saved Parameters. This is not possible as the modification of certain parameters trigger a restart request. It seams that at this Point the Parameter Loading fails or stops. Redoing loading after a reset did not help in my situation.

Currently I ended up with a partially loaded Parameter List. For example Accelerometer 0 was correctly initialised and Temperature Coefficients are loaded, but Gyro Data was missing as well as the rest of all TC Data (Gyro, Baro, etc).

I'm not sure if this is a QGC or PX4 Issue. QGC used is a daily build from 18.7.

This is a followup of https://github.com/PX4/Firmware/issues/7961 Details / Files / Screenshots are available there.

DonLakeFlyer, 2017-09-19:
This is due to optional parameters on the firmware side. QGC will only load parameters that it currently sees. You can workaround this by loading the parameter file multiple times until all the optional parameters become visible.

I think it's ok to load unknown parameters but I have to test to find out.

tops4u, 2017-09-19:
Repetitive Load did not solve the Problem. The opposite was the case after subsequent Loads some Parameters vanished again. At least multiple loads did not complete missing Params ie. Temp Compensation Params have only been loaded or the 1st Accel and no other Sensors.

DonLakeFlyer, 2020-11-20:
This has been fixed

closed 2020-11-20 by DonLakeFlyer