mavlink/qgroundcontrol #5681: Failed parameter loading when connecting multiple vehicles to QGC

Dnrodriguez374, 2017-09-20:
I am running QGC 3.2.4 Stable build. I have 5 different vehicles fitted with a Pixhawk autopilot that I need to connect to QGC via telemetry radio. 

Each vehicle has its own unique MAV_SYS_ID
Each vehicle has a default MAV_COMP_ID
There is a telemetry radio fitted onto each vehicle connected to the telem 1 port of the PX4 board
There is one telemetry radio connected to the control station running QGC

When I try to connect each vehicle individually, I have no issues. QGC will load the vehicle's full parameter list.

I let the connected vehicle and QGC settle. When I connect another vehicle, I see the vehicle connection on QGC and heartbeat is received, but loading parameters fails. 

Connecting the vehicles and receiving heartbeat is working, but 4 out of the 5 vehicles are essentially useless without their parameters loaded. I would like to be able to use mission protocol on each vehicle individually while they are connected.

I was able to connect two vehicles successfully one time with full parameters. I was able to upload missions individually to each vehicle connected. I have not been able to recreate that.

Here is a log of what happens when I try to connect a second vehicle to QGC.

[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCApplication.cc:259 - "Settings location "/home/daniel/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:99 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nMAVLinkProtocolLog.debug=true\n""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/Vehicle/MAVLinkLogManager.cc:359 - "MAVLink logs directory: "/home/daniel/Documents/QGroundControl/Logs""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QtLocationPlugin/QGCMapEngine.cpp:215 - "Map Cache in: "/home/daniel/.cache/QGCMapCache300" / "qgcMapCache.db""
[D] at qrc:///qml/QGroundControl/ScreenTools/ScreenTools.qml:117 - "0.05185185185185185"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/Controls/SliderSwitch.qml:74 - "47 56"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(0) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/MAVLinkProtocol.cc:413 - "Orphaned log file count 0"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/comm/MAVLinkProtocol.cc:381 - "Temp log "/tmp/FlightData759307.mavlink""
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(1) _vehicleArmed(0) guidedModeSupported(0) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(1) _vehicleArmed(0) guidedModeSupported(1) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(0) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(1) _vehicleArmed(0) guidedModeSupported(1) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(1) _vehiclePaused(0) _flightMode()"
[D] at qrc:///qml/QGroundControl/FlightDisplay/GuidedActionsController.qml:129 - "_activeVehicle(1) _vehicleArmed(0) guidedModeSupported(1) _vehicleFlying(0) _vehicleInRTLMode(0) pauseVehicleSupported(1) _vehiclePaused(0) _flightMode(Stabilized)"
[E] at qrc:/toolbar/GPSRTKIndicator.qml:108 - "qrc:/toolbar/GPSRTKIndicator.qml:108: ReferenceError: numSatValue is not defined"
[I] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:793 - "Parameters loaded from cache /home/daniel/.config/QGroundControl.org/ParamCache/4_1"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/MissionManager/MissionManager.cc:859 - "inProgressChanged"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:99 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nESP8266ComponentControllerLog.debug=true\nMAVLinkProtocolLog.debug=true\n""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/QGCLoggingCategory.cc:99 - "Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\nESP8266ComponentControllerLog.debug=true\nMAVLinkProtocolLog.debug=true\nParameterManagerLog.debug=true\n""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/MissionManager/MissionSettingsItem.cc:229 - "Setting home position QGeoCoordinate(30.3996, -84.2104, 10.596)"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:427 - ""V:5" Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1080 - ""V:5" Retrying initial parameter request list"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:427 - ""V:5" Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1080 - ""V:5" Retrying initial parameter request list"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:427 - ""V:5" Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1080 - ""V:5" Retrying initial parameter request list"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:427 - ""V:5" Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1080 - ""V:5" Retrying initial parameter request list"
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:427 - ""V:5" Request to refresh all parameters for component ID: "MAV_COMP_ID_ALL""
[D] at /home/travis/build/mavlink/qgroundcontrol/src/FactSystem/ParameterManager.cc:1087 - ""Vehicle 5 did not respond to request for parameters. This will cause QGroundControl to be unable to display its full user interface.""

DonLakeFlyer, 2018-06-20:
Is this still happening on latest daily build?

DonLakeFlyer, 2018-06-20:
Reopen if still happening

closed 2018-06-20 by DonLakeFlyer

Doolly, 2018-08-02:
I have this issue with daily build

DonLakeFlyer, 2018-08-02:
Pleasee enter full details of steps to reproduce and also turn on ParametereManagerLog and attach logs here.

dayL-W, 2019-05-14:
I have the same problem with QGC 3.5.0 and my connection method is the same as yours

dayL-W, 2019-05-14:
How did you solve it?

bys1123, 2019-05-14:
Looks like this still happen on 3.5.2

DonLakeFlyer, 2019-05-14:
Is this over a SiK Radio? If so, those have very limited bandwidth. It's more than likely that you are losing tons of packets since the comm pipe is full. You can see Mavlink loss rate from the Mavlink page in settings.

dayL-W, 2019-05-15:
> Looks like this still happen on 3.5.2

Yes, It still happen.

dayL-W, 2019-05-15:
> Is this over a SiK Radio? If so, those have very limited bandwidth. It's more than likely that you are losing tons of packets since the comm pipe is full. You can see Mavlink loss rate from the Mavlink page in settings.

Yes, I use SiK Radio. And all telemetries have the same net_id. Should I use WiFi Telemetry Radio?

dayL-W, 2019-05-15:
> Is this over a SiK Radio? If so, those have very limited bandwidth. It's more than likely that you are losing tons of packets since the comm pipe is full. You can see Mavlink loss rate from the Mavlink page in settings.

When I connect 2 vehicles, the packet loss rate of 2 vehicles is nearly 30%

DonLakeFlyer, 2019-05-15:
Yikes. That will never work. You're going to need to talk with some hardware people to figure out wht telemetry you need.

dayL-W, 2019-05-15:
> Yikes. That will never work. You're going to need to talk with some hardware people to figure out wht telemetry you need.

Thank you! I wonder how people connect multiple planes? Do they use multiple telemetry radios on GCS or just use only one telemetry which has high bandwidth?