mavlink/qgroundcontrol #5694: Plan: Structure Scan discussion

DonLakeFlyer, 2017-09-22:
This issue is for discussion of Structure Scan support in QGC. The simple concept from a functional standpoint is:
* Draw a polygon
* Specify the altitude for the first layer
* Specify the number of layers up you want to go from the first layer
* Specify the amount of height between each layer
* Specify a camera trigger distance to use while traversing the path
* Generate a mission which follows the path of the polygon for each layer triggering the images along the way

DonLakeFlyer, 2017-09-22:
Quick example:
[image]

lvale, 2017-09-22:
Hi @DonLakeFlyer 

Perhaps a desired picture resolution, and a picture target location

DonLakeFlyer, 2017-09-22:
> Perhaps a desired picture resolution, and a picture target location

I was thinking about resolution. If I think structure scan is just a vertical survey then I can consider the polygon to show to surface of the area to survey. Then if you specified a resolution the vehicle would back off the appropriate amount from the surface to get that resolution. But I'm not sure if that is how structure scan is used.

The picture target location would be perpendicular to the polygon edge facing in.

lvale, 2017-09-23:
what if you do a structure scan from the inside of a target?. Think survey the inside of the new apple building :)

birchera, 2017-09-23:
For concave corners the achievable angle of incident vs proximity to structure itself could be a factor. Not sure how to handle it though.

Also, I would think about having the option for a start and an end point. Otherwise it is limited to objects where you can fly around on all sides. Maybe a line would be better suited than a polygon.

DonLakeFlyer, 2017-09-23:
> what if you do a structure scan from the inside of a target?

> Also, I would think about having the option for a start and an end point.

Yup and yup, great stuff. Thanks guys.

DonLakeFlyer, 2017-09-27:
Functional support still missing:
* Circular scans

For discussion:
* Does it make sense to be able to set up structure scan like a vertical survey. With things like image resolution dictating distance from structure and so forth.
* Should there be support for video as well?
* Should there be support for running the gimbal angled down towards the structure?

lvale, 2017-09-27:
On a structure scan IMHO it makes sense to create a camera target path (that would take care of gimbal angles) and a vehicle path. By default user would only need to determine the vehicle path and the camera would be aligned with yaw. If the user would "detach" the camera target from the vehicle then the gimbal target location would be available to control 

Video support is a bonus, because apps that reconstruct from video also exist

The structure scan is a vertical survey :)

closed 2017-12-30 by DonLakeFlyer