mavlink/qgroundcontrol #5695: Plan: Corridor scan discussion

DonLakeFlyer, 2017-09-22:
This issue is for discussion of Corridor Scan support in QGC. The simple concept from a functional standpoint is:
* Draw a vector path which follows the center of the corridor you want to survey
* Specify the width of the corridor
* Specify the camera settings you want just like Survey
* Generate a mission which flies the vector path specified offseting multiple flights throught the path to get full coverage of the corridor width.

nfguide, 2017-09-25:
@DonLakeFlyer, you have nailed the definition perfectly.

There is an upcoming pipeline corridor project, which the right of way is 60m wide and 220km long.  So, yeah, this is needed, badly.  The guys at UgCS have a workaround which enables this functionality with UcGS planner.

Basically provide a way to input the line coordinates or consume the corridor data via .shp file or .kml/.kmz file for the center line.  Next, define the width of the corridor from the centerline, add altitude, camera info and overlap.  Calculate mission/flight plan, upload and go.

Looking forward to seeing this implemented, and happy to start testing.

Cheers!

Antiheavy, 2017-10-17:
I agree this described functionality is exactly how this should work.  I too just had an application for this last week mapping a road construction project.  Using "Survey" type waypoints was super painful and didn't really get the job done.  I am happy to help test on fixed wing when this functionality gets developed.

DonLakeFlyer, 2018-01-30:
WIP...
[image]

DonLakeFlyer, 2018-02-03:
Remaining work:
* Relative altitude support
* Turnaround
* Stats
* Greatest distance to and so forth

@cody303 Then an option for selecting maximum turn angle before requiring a turn around point. This way the aircraft could make small turns on the roads without having to make a new polygon to get the angle of the flight line correct.

I don't quite undestand this. Can you explain more.

DonLakeFlyer, 2018-02-03:
@Antiheavy Work on this has started.

Antiheavy, 2018-02-03:
> @Antiheavy Work on this has started.

it's looking good!   If I understand #cody303 right he is suggesting a special turn around maneuver for making tight corners in a corridor,  I can see that being useful for fixed wing anyway, e.g. something like this maybe:
[image]

DonLakeFlyer, 2018-02-03:
Ah!. Hmm, that's going to be pretty tricky to implement.

DonLakeFlyer, 2018-02-03:
Damn fixed wing guys making things harder :)

Antiheavy, 2018-02-03:
> that's going to be pretty tricky to implement

i agree, tricky.  I think there is value in pushing something out for testing before you try to tackle this extra tight corner turnaround feature.  It might be that it works "good enough" without it (my gut tells me there is a 40% chance it might be good enough without it).

When/if it comes time to implementing, you could do it with 2-3 waypoints to make a pattern or 1 off-set Loiter waypoint with tangent exit criteria.  Might need both an angle and a turn radius value to calculate properly for different airframes.

nfguide, 2018-02-03:
Agree with @Antiheavy on the turn points.

One option would be to extend the initial leg beyond the transition, add another turn to wp, then line up onto the next leg.  

As noted in this pic.
[image]

DonLakeFlyer, 2018-02-03:
Can folks comment on the "Take images in turnaround" as seen in Survey? Are there really cases where turning this on is valuable? Trying to simplify things with all the Pattern type.

@sundanceMediagroup FYI

cody303, 2018-02-03:
Instead of turning around and continuing it might be easier and possibly more efficient to break the pattern into two patterns at the turn. Fly one side first then the other side with some overlap in between the two areas.

Antiheavy, 2018-02-03:
> Instead of turning around and continuing it might be easier and possibly more efficient to break the pattern into two patterns at the turn. Fly one side first then the other side with some overlap in between the two areas.

I agree, this is one way we might get by without a special turn around on tight corners.

Antiheavy, 2018-02-03:
> Can folks comment on the "Take images in turnaround" as seen in Survey?

I find "Take images in turnaround" selection option very useful and important for the Survey pattern, especially for fixed wing.  If this unfinished PR ever gets in (https://github.com/PX4/Firmware/pull/7739), then "Take images in turnaround" will be slightly less important for Survey, although I will still advocate strongly to keep it.

I think "Take images in the turnarounds" is **probably** much less important for Corridor scan and could be defaulted to not taking pictures in the turnaround.  If https://github.com/PX4/Firmware/pull/7739 gets implemented effectively then I think you could hide the option for Corridor scan.  Of course if you leave the option in for Survey and don't have it for Corridor people will probably ask for it anyway...

Another thing to consider is if you want to treat MC, VTOL, and FW differently for this feature.  I dunno.

I am ALL FOR simplification, streamlining, and ease of use, but "Taking pictures in the turnarounds" is still a required option for some of us (in Survey at least).

nfguide, 2018-02-04:
@DonLakeFlyer 

Shooting images in turns, at least from a corridor mapping perspective is useless, unless the camera is on a gimbal.  If that is the case, the multi point may not be needed.

On a 30km segment, having multiple missions would be a waste, and manually setting waypoints at changes in direction would be the way to go.  Currently doing so with UgCS today, and future releases will have this baked into the app.

EBee and Sensefly have nailed it and not quite sure how it was implemented.

My $.02 worth.  :)

nfguide, 2018-02-04:
Just for some perspective of a 30km segment, with turn points called out, and zoomed in and noted in the post above.

[image]

DonLakeFlyer, 2018-02-04:
> I find "Take images in turnaround" selection option very useful and important for the Survey pattern,

@Antiheavy Thanks for info.

DonLakeFlyer, 2018-02-04:
Remaining work:
* Take images in turnaround
* Fixed Wing max turn radius support
* Update altitude when alt change from Mission Start

Another interesting thing is with the refactoring of common code the combination of Corridor Scan and Structure Scan code generic code gives me all I need for a vertical surface scan based on a polyline (instead of a polygon like Structure Scan).

DonLakeFlyer, 2018-02-04:
Also just FYI: This work is what I would call Phase I. In March I plan to start Phase II Corridor Scan which will include terrain following, with maximum flight gradient adjustment. Then I can take all that terrain code and apply back everthing else (Survey, generic mission, ...)

Antiheavy, 2018-02-04:
> take all that terrain code and apply back to everthing else (Survey, generic mission, ...)

exciting!

Antiheavy, 2018-02-04:
someone mentioned eBee. Looks like it creates multiple long, skinny, and straight "survey" waypoints and overlap them where the corridor changes direction.  By forcing odd number of passes the vehicle is always on the right end to start the next segment.

screengrab from Sensefly web site:
[image]

One upside to this approach is the area on the ground at the turn often would like a little extra wide photo coverage anyway.

So:
One approach might be for a variable called "minimum turn angle for new segment" (or something) that breaks the mission into multiple overlapping corridor patterns and forces odd number of passes.  Don't force odd number of passes on single segment corridors though since people might like to end where they started in that case.

nfguide, 2018-02-04:
With short and bendy corridors, the above approach makes sense, using odd number of flight lines and landing at the start of the next section.

On 30km segments, that is not really a viable option, as we need even number flight lines to always RTL for the next 30km segment.

Definitely need a minimum turn angle then set the wp's for the next leg.

Antiheavy, 2018-03-10:
> On 30km segments, that is not really a viable option, as we need even number flight lines to always RTL for the next 30km segment.

It seems like a selector for "even", "odd", or "any" number of passes would be very useful here.

closed 2018-06-20 by DonLakeFlyer