mavlink/qgroundcontrol #5697: PX4 hide params that aren't user set

dagar, 2017-09-23:
In PX4 all calibration data is exposed as parameters (several hundred), but they're set via calibration routines and not intended to be edited by a user. In some cases these are in the same group as parameters that are user configurable. Hiding calibration params in the QGC param editor tree would be a nice simplification that makes it easier to find the params a user should actually be able to change and prevent accidents.

@DonLakeFlyer what do you think about adding a tag like `@advanced` or `@calibration` and hiding these by default with an option to show them?

The first pass would include all calibration params except things like external compass rotation intended to be, thermal calibration params, RC calibration params.

This is a version of @sanderux's advanced param idea that has come up a few times.

hamishwillee, 2017-09-23:
OR `@internal`

sanderux, 2017-09-24:
I would propose @level to keep it flexible for additional values. We could start with
- BASIC: a user may change these
- ADVANCED: change these for new airframe configs
- EXPERT: calib, etc

dagar, 2017-09-24:
@sanderux I think first making a distinction between params calculated and set vehicle side and params set on the GCS by a user is important. When we previously discussed BASIC/ADVANCED I couldn't think of a good example of something BASIC that isn't already broken out as a nice configuration screen in QGC. If you can maybe we should focus on creating more configuration tabs, that code is actually quite approachable. https://github.com/mavlink/qgroundcontrol/blob/master/src/AutoPilotPlugins/PX4/CameraComponent.qml#L101

Really I think those are the levels. 
  - Nice configuration tabs (that ultimately set parameters)
  - param tree (not including anything set autopilot side)
  - Complete param tree

hamishwillee, 2017-09-24:
> I think first making a distinction between params calculated and set vehicle side and params set on the GCS by a user is important

This is my concern . Hence `@internal` or `@autopilot_only`. I guess you could call this expert if you liked, but there is probably a distinction between things you might want a real expert to touch and things that even an expert should not be varying through GCS.

DonLakeFlyer, 2017-09-24:
I think something would be good here. But I"m not sure the categorizing is based on the complexity of the param. My personal view is that most users should never use the parameter editor. If that is not the case then something is wrong, and some sort of vehicle setup ui is missing from the product. To the categorization would be more of a grouping to help you find what you are looking for. Not something like advanced. For example here are the set of parameters which come into play when flying a mission.

@sanderux I'd be interested to here why you spend time in the parameter editor?

dagar, 2017-09-24:
@hamishwillee the other thing we could consider on the PX4 side is not even exposing them as regular params, but then you lose a potential backup mechanism.

sanderux, 2017-09-25:
@donlakeflyer i'm not sure as a px4 dev i would be a good example, but most vtol params are not exposed in any other way. 

But dor example: the DeltaQuad that we produce does not use an airspeed sensor so we set cruise throttle.
Also quadchute sometimes needs tuning based on the mission.

DonLakeFlyer, 2018-01-03:
https://github.com/PX4/Firmware/issues/8568

DonLakeFlyer, 2018-01-04:
Replace by #5965

closed 2018-01-04 by DonLakeFlyer