mavlink/qgroundcontrol #5704: Geotagging causes program crash

JonReacher, 2017-09-26:
Using the geotag feature in either stable 3.2.4 or daily build causes the program to crash after geotagging. It also won't tag all of the images in the folder.

DonLakeFlyer, 2017-09-26:
@birchera Could you take a look?

birchera, 2017-09-26:
Does this happen consistently? Or only on one dataset?

How many images do you have to tag? And do you know how many trigger events you have logged? And how many images get tagged before the crash?

JonReacher, 2017-09-26:
I tried my first set of images, 156 photos. I hadn't tried this before, but it seemed pretty straight forward. It crashed before tagging all of the photos. It may be an issue with some untaken/missed images vs the number of commanded triggers.
 Since I thought that may be the case, I added some additional unrelated photos to the folder just to match the number of triggers. It still wouldn't make it through all of the photos before crashing.

birchera, 2017-09-26:
Missing images should only lead to wrong tagging, but not to a crash. There is a [check](https://github.com/mavlink/qgroundcontrol/blob/master/src/AnalyzeView/GeoTagController.cc#L259-L260) on the array index.

If you enable GeotaggingLog topic does that tell you something?

JonReacher, 2017-09-26:
> If you enable GeotaggingLog topic does that tell you something?

That's where you start to lose me. How do I enable GeotaggingLog topic?

birchera, 2017-09-26:
If you go to the console tab and click on set logging. Then you can chose the category.

Do you happen to have another dataset to see whether it is reproducible?

What firmware and what kind of log do you use?

JonReacher, 2017-09-26:
I just got another set. I'll try again and report. I'm using PX4 1.6.5 stable and it produces a ulog.

JonReacher, 2017-09-26:
This time it seemed to work fine. I had missed images, so fewer photos than triggers, and it just gave the spinning wheel after it tagged what it could, but it didn't crash. I also tried it with the same number of photos as triggers and it did fine. So I don't know what caused it, but I didn't get it to reproduce the effect.

JonReacher, 2017-10-18:
This definitely seems to be hit or miss on whether it will work. I haven't ever gotten it to geotag a complete missions worth of photos. That's with no missed photos, same number of triggers as taken photos. It consistently doesn't tag a handful of photos. Sometimes it crashes, sometimes not, but never tags them all.

 I've resorted to trying once, counting the number of photos not tagged, and then manually adjusting everything (adding or deleting photos) to match the number of photos in the folder to the number of photos it is tagging. Any thoughts?

DonLakeFlyer, 2018-04-30:
Should be fixed with pulls linked to #6234

closed 2018-04-30 by DonLakeFlyer