mavlink/qgroundcontrol #5710: Not able to add ArcGIS Maptype in Q ground control

lakshaydang, 2017-09-28:
How to add the GIS map in QGC. Also there are other Esri maps but i am not able to build it.
Anyone can help me with this?

DonLakeFlyer, 2017-09-28:
Support for ESRI is not fully plumbed through QGC.

DonLakeFlyer, 2017-09-28:
@dogmaphobic The token is there, but you can't pick the map type right?

lakshaydang, 2017-09-28:
yes, i was trying for it but qt shows the error. Do you know how to do that i am not able to do that.
Also there is one ArcGIS sdk, can we integrate that with the qt?

dogmaphobic, 2017-09-28:
You need your own ESRI account and your own token. Just like MapBox. With that said, QGC only supports "tile" maps. There is no support for the GIS api.

lakshaydang, 2017-09-29:
So any way is there how can i integrate it or you know any person who can help me with that.
Also is there any work going on for GIS integration in QGC and upto when it will be done?

dogmaphobic, 2017-09-29:
Forgot to mention, all tiles are bitmaps. Either PNG or Jpeg and usually 256x256 or 512x512 for high density displays (QGC uses 512x512).

lakshaydang, 2017-09-29:
Ok there is an app in Qt i.e QGIS any how we can use that with q ground control ?

dogmaphobic, 2017-09-29:
QGIS is a desktop application. It uses a rather different API and it won't ever work on mobile. I would imagine there could be a way to make both applications interoperable so each can do what they are supposed to do best. It all depends what the needs are, which I don't know.

kikislater, 2017-09-29:
Qgis uses gdal/ogr as a data driver. Gdal is now available in Mission planner. May be it's a good way to implement gdal in QGC.... Don't know if it works under mobile platform

A first good option IMO, would be kml/kmz support because most GIS users know how to convert various GIS data to kml. But here it's possible to save kml, why not open ? : https://github.com/mavlink/qgroundcontrol/pull/5499

@lakshaydang : Which format is it ? In GIS we have geometry data file, not map ! Map is a final product. Why want you use proprietary format whereas there is a standard exchange format (still proprietary but documented) : shapefile ?

closed 2025-05-14 by HTRamsey