mavlink/qgroundcontrol #5711: Eachine ROTG01 UVC FPV Receiver not selectable / Android

fingadar, 2017-09-28:
The Eachine ROTG01 UVC FPV receiver is cheap and simple, see https://www.banggood.com/Eachine-ROTG01-UVC-OTG-5_8G-150CH-Full-Channel-FPV-Receiver-For-Android-Mobile-Phone-Smartphone-p-1147692.html for more information.
The receiver itself is recognized by Android (it even suggests qGoundControl as standard application when connecting the ROTG01 receiver to the tablet). But in qGroundControl, the UVC device does not appear in the list of video sources.
The receiver itself works well in other apps (eg. Tower app V.4). 
Tested with two different tablets (Nexus 7 with Android 4 and Medion x10302 with Android 6) with same results.

If you need any further informations or anything other that I can do, please tell me :)

Best regards,
Stefan

DonLakeFlyer, 2017-09-29:
@dogmaphobic Is this supposed to be working on Android?

dogmaphobic, 2017-09-29:
I don't remember. The idea is that if the UVC device shows up as a regular *Camera* to Qt, it will work. I'm not sure Qt supports external cameras under Android.

taileron, 2017-10-05:
ROTG01 shows up as USB 2.0 camera in all platforms of Qgroundcontrol (Mac/Win/Ubuntu) expect Android but in Android lots of other Apps connect to ROTG01 with super low latency (e.g. with Windows the latency is way too high) so Android would be the preferred platform if it would work in future ...

taileron, 2017-10-11:
... sorry in Ubuntu 16.04 QGC behaves the same as in Android ROTG01 isn´t available in the selection but all other tools can use it e.g. "cheese"

taileron, 2019-03-14:
current best Workaround is this converter composite to rtsp: Vivotek VS8100-V2 works perfect out of the box (rtsp://192.168.x.x:554/live.sdp), converts analog video (from a 5.8G FPV receiver) into a rtsp stream that can be displayed by QGC with quite similar latency to ROTG0X but at considerably better image quality (image parameters and deinterlacing configurable by web-interface) 

second workaround I´ve done -> installing osx86 to the tablet and use the Mac version of QGC running fullscreen and with a bit lower latency than with the other platforms while being able to do screen-record with 30 fps

closed 2026-10-04 by github-actions[bot]