mavlink/qgroundcontrol #5718: TTS under Linux

dogmaphobic, 2017-09-29:
The main `README.md` talks about some extra speech libraries. Are these still needed with the new Qt TTS module?

```
Ubuntu: sudo apt-get install espeak libespeak-dev libudev-dev libsdl2-dev
Fedora: sudo dnf install espeak espeak-devel SDL2-devel SDL2 systemd-devel
Arch Linux: pacman -Sy espeak
```

DonLakeFlyer, 2017-09-29:
Can you uninstall from Linux and see what happens?

kikislater, 2017-09-29:
On latest daily build under archlinux, I need to install qt5-speech to make build working.
pacman -S qt5-speech

If qt5-speech is not present, the following error occurs and program won't open : 
```
$ ./QGroundControl.AppImage 
/tmp/.mount_Q8rwvO/QGroundControl: error while loading shared libraries: libQt5TextToSpeech.so.5: cannot open shared object file: No such file or directory
```

dogmaphobic, 2017-09-29:
Ok, this is the opposite of my original question. I don't have any functional Linux VM right now to test this. The new code uses Qt's own TTS engine. The original question was if we still need the old libraries used by the old code. It now seems Linux installs need yet another bit of its Qt puzzle installed to provide said speech engine. Why is Qt on Linux all piecemeal like this?

dogmaphobic, 2017-09-30:
Right, that's what I meant about piecemeal. For Linux, all needed libraries must be explicitly defined here:

https://github.com/mavlink/qgroundcontrol/blob/master/QGCSetup.pri#L104-L125

Other platforms seem to have *deployment* tools that do this automatically.

DonLakeFlyer, 2017-09-30:
Ah, sorry I wasn't paying attention. I thought this was a building problem.

DonLakeFlyer, 2017-09-30:
I tried building this, but I don't see where the new Qt stuff is used. All the old text to speech stuff is still in there?

DonLakeFlyer, 2017-09-30:
```
void QGCAudioWorker::say(QString inText)
{
#ifdef __android__
    Q_UNUSED(inText);
#else
    static bool threadInit = false;
    if (!threadInit) {
        threadInit = true;
        init();
    }

    if (!muted)
    {
        QString text = fixTextMessageForAudio(inText);

#if defined _MSC_VER && defined QGC_SPEECH_ENABLED
        HRESULT hr = pVoice->Speak(text.toStdWString().c_str(), SPF_DEFAULT, NULL);
        if (FAILED(hr)) {
            qDebug() << "Speak failed, HR:" << QString("%1").arg(hr, 0, 16);
        }
#elif defined Q_OS_LINUX && defined QGC_SPEECH_ENABLED
        // Set size of string for espeak: +1 for the null-character
        unsigned int espeak_size = strlen(text.toStdString().c_str()) + 1;
        espeak_Synth(text.toStdString().c_str(), espeak_size, 0, POS_CHARACTER, 0, espeakCHARS_AUTO, NULL, NULL);
        espeak_Synchronize();
#elif (defined __macos__) && defined QGC_SPEECH_ENABLED
        macSpeak(text.toStdString().c_str());
#elif (defined __ios__) && defined QGC_SPEECH_ENABLED
        iOSSpeak(text);
#else
        // Make sure there isn't an unused variable warning when speech output is disabled
        Q_UNUSED(inText);
#endif
    }
#endif // __android__
```

dogmaphobic, 2017-09-30:
You probably have not pulled current master:

https://github.com/mavlink/qgroundcontrol/blob/master/src/GAudioOutput.cc

`QGCAudioWorked` no longer exists.

DonLakeFlyer, 2017-09-30:
Huh, could have sworn I did that. I'll try again and figure out what needs to happen.

kikislater, 2017-09-30:
so I don't understand, is it a missing library or a normal behaviour ? feel free to tell us when it will be updated, I will test.

DonLakeFlyer, 2017-09-30:
qt5-speech is a missing library. The question is also whether espeak libespeak-dev  are still needed. Which I haven't figured out yet.

DonLakeFlyer, 2017-10-30:
I just fixed up the Linux build so it boots again. I then tested Daily build installed to fresh 16.04 vm. I get the no plugin error. @dogmaphobic Can you figure out how to get TTS working again on Linux?

DonLakeFlyer, 2017-10-31:
Qt boom: https://bugreports.qt.io/browse/QTBUG-58461

pritamghanghas, 2017-10-31:
Qt5.9.2 has the fix. Test it on ubuntu 17.04
On ubuntu, all you need is 
`sudo apt-get install speech-dispatcher`
it will take care of all the dependencies. 
If you doubt your ubuntu installation run following command to check speech
`spd-say hello`

DonLakeFlyer, 2018-01-12:
Can Linux folks try the new daily AppImage builds? Should be running 5.9.3 now with speech working I hope. Possibly some additional things still missing.

DonLakeFlyer, 2018-01-13:
Closing until I hear not working.

closed 2018-01-13 by DonLakeFlyer