mavlink/qgroundcontrol #5721: lack of support for Raspberry PI cam

Mocart96, 2017-10-01:
For video streaming from the Raspberry PI ( pi2, pi3, pi0w ) you can use PI cam, the best solution for low latency streaming. In the case of Pi cam great results you can achive running following pipeline from pi shell:
`raspivid -n -w 864 -h 480 -b 750000 -fps 25 -t 0 -o - | \
gst-launch-1.0 --gst-debug-level=0 -v \
fdsrc ! \
h264parse ! \
rtph264pay config-interval=10 pt=96 ! \
udpsink host=xxx.xxx.xxx.xxx port=5600`
(resolution, fps, port, destination can be defined by user)
this pipeline can be well decoded (with low latency ) on the ground station side with a following function:
`c:/gstreamer/1.0/x86_64/bin/gst-launch-1.0.exe -e -v udpsrc port=5600 ! application/x-rtp, payload=96 ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! fpsdisplaysink sync=false text-overlay=false`
Can you implement option to input custom pipeline under UDP option to allow imput following command for Raspberry users.
`udpsrc port=5600 ! application/x-rtp, payload=96 ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! fpsdisplaysink sync=false text-overlay=false`
I did test it with different android sofware and it works perfect. With the current decoding instraction I can see video feed working but very often it shows pixels on the screen. My suggestion is related to android application of course.

baomin, 2017-11-29:
good !!!

baomin, 2017-12-04:
raspivid -n -w 864 -h 480 -b 750000 -fps 25 -t 0 -o - | gst-launch-1.0 --gst-debug-level=0 -v fdsrc ! h264parse ! rtph264pay config-interval=10 pt=96 ! udpsink host=xxx.xxx.xxx.xxx port=5600

Sinamics, 2019-08-23:
Hi
Would appreciate if we could get the option to add custom pipeline. 
In some cases when streaming on lower bandwidth or through several layers of NAT firewalls using NAT traversal, the video image gets greyed out.
Mission Planner also had this issue until Michael Osborne added the option for custom commands. 
The grey overlay issue can be solved by adding **rtpjitterbuffer** to the pipeline.

> This element reorders and removes duplicate RTP packets as they are received from a network source

[image]

FYI
For Mission planner this can be solved by adding this line as gstreamer source. 
`udpsrc port=5600 caps = "application/x-rtp, media=video, clock-rate=90000, encoding-name=H264, payload=96" ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert ! video/x-raw,format=BGRA ! appsink name=outsink`

More info on rtpjitterbuffer can be found here.
https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gst-plugins-good/html/gst-plugins-good-plugins-rtpjitterbuffer.html

Thank you in advance.

ask21on, 2019-08-29:
Is there any news about this?
It seems my video suffer from this issue too...

closed 2026-10-04 by github-actions[bot]