mavlink/qgroundcontrol #5724: Unable to configure if parameters incomplete

AndKe, 2017-10-03:
There is a lengthy warning about configuration screens not being available if parameter fetch is incomplete.

Even going to full parameter list, and refetching does not help.
The application must be shutdown, then started. 

please add a check after a successful parameter refresh.

DonLakeFlyer, 2017-10-05:
This isn't really fixable without some major architectural changes. Once you miss getting parameters the first time you are dead in the water since there is no mechanism architecturally to signal that they showed up on a subsequent try. Would be nice if there was but not going to be high priority to fix.

DonLakeFlyer, 2017-10-11:
I'm going to leave this handing around. It's good to remember this is a problem. Hard to fix. but still would be good to take care of at some point.

closed 2026-10-04 by github-actions[bot]