mavlink/qgroundcontrol #5725: RTL: Inconsistent naming

sanderux, 2017-10-03:
I propose we consistently refer to RTL as Return To Launch. 

[image]

DonLakeFlyer, 2017-10-05:
Combo box values come from firmware parameter meta data. So that requires a firmware change. Happy to change to anything if you can sync up what firmware wants to call it.

dagar, 2017-10-08:
For now it should probably be "return to land" everywhere, although it may be worth distinguishing "return to home" and "return to land" soon with all these other RTL variations (RTL -> mission landing, smart RTL, RTL with safe points).

hamishwillee, 2017-10-08:
>  although it may be worth distinguishing "return to home" and "return to land" soon with all these other RTL variations (RTL -> mission landing, smart RTL, RTL with safe points).

Yes if these are different "modes" and no if they are the same mode with some parameter that define where the land point is. We could consider something like "Safe Land" mode as well to keep this generic. 
For docs we are currently standardising against what QGC says in the mode selection boxes - so it is "Return" mode.

sanderux, 2017-10-09:
> For now it should probably be "return to land" everywhere
What does that mean? I don't even know if that's correct English

I would propose 3 modes, or the latter 2 as options on the first
Return To Launch (RTL)
Smart Return To Launch (SRTL)
Fail Safe Land (FSL) (Rally point, Home or Destination, which ever is closest)

Antiheavy, 2017-10-09:
oh my favorite topic!  Yeah, the RTL naming is inconsistent all over the place within the various ground stations and firmwares and documentation. 

Fixed Wing and VTOL are sort of fundamentally different with how they do/should work.  IMHO Return to Launch isn't the right term to use for a number of reasons.  I think "Return to Home" (RTH) and "Return to Land" (RTL) are the better and more commonly used terms in the industry.

**Return to Land**
Traditionally (going back to before the advent of quad copters et al) RTL has generally, and most often, stood for Return to Land.  This term is somewhat fixed wing focused, but that was all we had to consider back then (we had helicopters, but they called it Return to Land too).  There was also a concept of Return to Home, but that involved loitering (circling or hovering) above the Home point and typically was not associated with actually landing.

Return to Land should be associated with returning to a pre-programmed "landing" location which might be a "Land" waypoint or pattern of some type.

**Return to Home**
Now that VTOL multicopters are the thing the concept of Return to Home (RTH) has become the defacto standard behavior and terminology for VTOL vehicles returning to the indented landing spot.  VTOLs simplified the whole thing by combining the Home point with the default land point and having it automatically set upon launch.  Ingenious really!  However, thus spawns our confusion.

Return to Home actually best describes what the autopilot is actually doing.  Most commercially sold systems update the Home point upon takeoff and inform the user that this is happening.  So while "Return to Launch" is effectively what is going to happen, it isn't what the autopilot is actually thinking it is doing.  Also many systems allow the user to change the Home point during flight, so a Return would no longer go to the Launch point.

**Advanced behaviors**
advanced behaviors such as the smart return and rally points or other topics, well, that adds another dimension to the discussion and doesn't necessarily have established industry norms to follow yet.

hamishwillee, 2017-10-09:
>> For now it should probably be "return to land" everywhere

> What does that mean? I don't even know if that's correct English

@sanderux 
- This is correct English but it is ambiguous. It means either got back to be over land (e.g. from the sea) or it might mean "return with intention of landing". In either case there is no indication of where you are returning to (home, launch, or safe point). 
- Return and land is closer to what we are after, but again, loses the destination landing point. 
- Currently QGC refers to this as "Return" which is (coincidentally) the most accurate, since it doesn't imply where we return, and it doesn't imply landing (which we don't necessarily do based on parameters). 

@Antiheavy Given the inconsistency of naming conventions I wouldn't be too tied up about the historical naming conventions - though if we end up with a mode that returns to home then we'd be silly not to use that name. 

I'd suggest that PX4 decide all the return behaviours we want and how they are configured (either as variations of a single mode or as unique modes) and then we work out good names from that. Lots of options - Return to Base, Return, Travel to Safe Zone, Safe Recover, etc. (Recover not bad, since it covers both home, launch or safe points)

tubeme, 2017-10-30:
I like the @hamishwillee "Safe Land" when landing on alternative spot.
The other could just stay "Return" with a warning message if it is Failsafe Return. And "Smart Return" the @sanderux distinction. We don't have to make it more complicated. The simpler the better in the pilot's cockpit if I can say it this way. Voice messaging will announce if it is Failsafe Return as well or smart return or safe land.

- Return (Comanded or Failsafe)
- Smart Return
- Safe Land

Antiheavy, 2017-10-30:
I think this can be viewed even more simply in that "normal" Return and "smart" Return are really the same thing from a user perspective, it is just "normal" return climbs to a return alt to avoid potential obstacles and "smart" return follows a pre-flown path back to avoid potential obstacles.  The user will have selected one of the options from the params and will expect that behavior.  So from QGC standpoint just the word "Return" is good enough for both.

"Safe Land" or Rally or whatever if finally is called really should be a separate mode and not part of Return.

DonLakeFlyer, 2019-12-24:
This has changed to "Launch".

closed 2019-12-24 by DonLakeFlyer

Antiheavy, 2019-12-24:
I think it is still just called Return Mode.