mavlink/qgroundcontrol #5730: **Feature request** improvements to replay function

Antiheavy, 2017-10-05:
Replay is a VERY useful function to have.  The current implementation is too basic.  These two improvements would make it much more useable:
- Ability to play forward and play backward.
- Ability to play forward/backward and faster/slower speeds.

Here is a couple concepts of how that might look:
[image]

somewhat related to:
https://github.com/mavlink/qgroundcontrol/issues/4268
https://github.com/mavlink/qgroundcontrol/issues/3601
https://github.com/mavlink/qgroundcontrol/issues/3375

fredrikfalkman, 2017-10-21:
Also, the ability to enter a specific time would be nice. And of course, live scrubbing!

Antiheavy, 2019-08-15:
> , live scrubbing!

live scrubbing works somewhat better in recent builds of QGC.

Antiheavy, 2019-08-15:
It was suggested to provide more context to this request and specific things people are trying to do with replay.  Here are some more thoughts:

1) **Case of a lost vehicle:** For purposes of trying to find the vehicle it is very helpful to know last known: location, altitude, ground course, speeds, and attitude (angles).  A print out of these things could be helpful, but a replay that allows easy forward/reverse and slow/fast playback can really help reconstruct the final moments before the vehicle was lost to determine both what cause the vehicle to be lost and also how large the search area might be.

2) **Analyze widget playback:**  sometimes we'd like to playback certain telemetry items and watch them slower or faster.  Often we might want to replay a large section of log back very quickly to build out those strip charts faster.

Antiheavy, 2019-08-28:
I think I can close this issues as it is generally address by: https://github.com/mavlink/qgroundcontrol/pull/7686

There are still some headaches, but I've opened them as stand alone issues:
https://github.com/mavlink/qgroundcontrol/issues/7710
https://github.com/mavlink/qgroundcontrol/issues/7709
https://github.com/mavlink/qgroundcontrol/issues/7708

closed 2019-08-28 by Antiheavy