mavlink/qgroundcontrol #5736: Announce battery lower than: XX% only works during the first flight

Antiheavy, 2017-10-10:
Tested in SITL using PX4 fixed wing.  QGC function to "Announce battery lower than: xx%" only works during the first flight during that QGC application session.  subsequent flights do not have the QGC announcements when the battery goes below this value.

[image]

DonLakeFlyer, 2017-10-26:
From looking at the code it looks like this can happen during the resume mission pattern. It only happens if you turn off the vehicle, but do not disconnected QGC and then turn the vehicle back on. Is that your case?

closed 2017-10-26 by DonLakeFlyer

Antiheavy, 2017-10-26:
> It only happens if you turn off the vehicle, but do not disconnected QGC and then turn the vehicle back on. Is that your case?

that is exactly how I was testing it.  I didn't try to "disconnect" QGC first.  This is how we typically operate even if not using "resume mission" function because we may be flying multiple other flights at the same location, so we don't bother to "disconnect" QGC in-between flights.