mavlink/qgroundcontrol #5737: Battery swap and mission resume procedure is unclear.

AndKe, 2017-10-10:
So, I get that the intention is to crop a mission, after landing for battery swap, but it's still unclear how it's supposed to be done.  (I did some testing)

I was flying a mission, (ArduCopter in auto mode) - decided battery needed to be changed in the middle  between waypoints that are 500m apart.

If I switch mode via RC to RTL, or anything manual - camera keeps on trigging by distance.
Apparently same happens if I pause mission, then RTL.

How is it supposed to work/ be used ?  Some smart behavior can be added to stop trigging by distance when mode changes, or maybe a dedicated "come for battery swap" could be added, then, the mission would be backed up not to previous WP, but rather "stopped_position - 4*trigging_distance" ?

DonLakeFlyer, 2017-10-11:
> If I switch mode via RC to RTL, or anything manual - camera keeps on trigging by distance.

That would require firmware support. At that point QGC doesn't really know what is going on. Vehicle is in control of running missions.

AndKe, 2017-10-11:
Well, it could require firmware support, I agree that would be the easy way.

QGC could, detect mode change, send and verify DO_SET_CAM_TRIGG_DIST = 0 , (restore it if mode is changed back to AUTO)   
store/restore is important if for example loitering fixed wing. 

Example; a mission is at some point to enter CTR, but is not yet cleared to do so (due to traffic)  - the plane would usually be put in a circle, but we do not wish 500photos to be taken while loitering. 

So one way is to handle it using QGC, to detect mode change and stop trigger.  Another, is to implement that into a "pause" feature, where a mission is paused, and cam trig distance is temporarily set to 0 until unpaused.  (So it would stay "paused", when RTL'ed for battery swap.)

In any case, we need to think of a proper way to use the feature, and facilitate for battery changes both before change, and after change (taken care of by the resume feature.)