mavlink/qgroundcontrol #5739: Android: Load mission .plans  from outside app internal storage

RPSpicer, 2017-10-11:
I have written a piece of code that generates valid .plan files for desktop QGroundControl, but would like to load these files on an Android phone or tablet. The current configuration of the QGroundControl app only reads and writes .plans within the app's private/internal storage. 

I would like to add the capability to read files from arbitrary URIs  on device shared storage.  

My ideal use case would be to add an IntentFilter to QGC that takes the URI to a .plan file in shared storage and opens it in the app. This avoids the need to implement a more complex file picker in QGC, but still makes the functionality available to external developers.

If I were to implement either/or and submit a PR, is this a thing the team would consider accepting?

DonLakeFlyer, 2017-10-11:
Why can't you just write to files to the QGC location?

pmtolk, 2017-10-11:
That is what we would like to accomplish. In my brief research into this issue it requires a android:sharedUserId in the manifest 
https://stackoverflow.com/questions/9783765/what-is-shareduserid-in-android-and-how-is-it-used

DonLakeFlyer, 2017-10-11:
@dogmaphobic Do you happen to know anything about this? I had thought anyone could write to that location. If not maybe we could set something on the app to allow anyone to read/write to the QGC files location.

jaxxzer, 2017-10-11:
Can't you just load/save mission files wherever? (Click the Sync button)

jaxxzer, 2017-10-11:
> Can't you just load/save mission files wherever? (Click the Sync button)

I see this isn't the case on mobile.

jaxxzer, 2017-10-11:
The missions are loaded from the 'Missions' directory under the 'File Save Path' in the miscellaneous section of the _General_ tab on the App Settings page.

The option to change this is shown on Android, but it can't be changed after clicking it.

`/storage/emulated/0/QGroundControl/Missions/`

DonLakeFlyer, 2017-10-17:
> The option to change this is shown on Android, but it can't be changed after clicking it.

That's a bug. That option shouldn't show up. It's a fixed location on mobile builds.

DonLakeFlyer, 2017-10-17:
Are you sure you can't write to the location? It's supposed to be writable. For example My Files app can see it and modify.

pmtolk, 2017-10-18:
I have tested and it is working, here is a complete example
```java
public static void SaveToFile(String fileContents, String fileName, Context applicationContext){
        File dir2Save = new File(Environment.getExternalStorageDirectory() + "/QGroundControl/Missions");
        String extStorageState = Environment.getExternalStorageState(dir2Save);
        if (Environment.MEDIA_MOUNTED.equals(extStorageState)) {
            Log.v(TAG, "Media is mounted!");
            FileOutputStream fOut;
            OutputStreamWriter fWriter;
            try {
                File fpfFile = new File(dir2Save, fileName);
                Log.v(TAG, "fpfFile2: " + fpfFile);
                if (!dir2Save.mkdirs()) {
                    Log.v(TAG, "Directory2 " + dir2Save + " exists, not creating.");
                }
                if (!fpfFile.createNewFile()) {
                    Log.e(TAG, "Error creating new file for flightplan: " + fpfFile);
                }
                fOut = new FileOutputStream(fpfFile);
                fWriter = new OutputStreamWriter(fOut);
                fWriter.write(fileContents);
                fWriter.close();
                fOut.close();
                //need this so windows explorer can see file for some reason
                Intent intent = new Intent(Intent.ACTION_MEDIA_SCANNER_SCAN_FILE);
                intent.setData(Uri.fromFile(fpfFile));
                applicationContext.sendBroadcast(intent);

            } catch (IOException ex) {
                Log.e(TAG, Log.getStackTraceString(ex));
            } catch (Exception ex) {
                Log.e(TAG, ex.getMessage());
            }
        }else {
            Log.w(TAG, "Media cannot be mounted! Skipping save file");
        }
    }
```

DonLakeFlyer, 2017-10-18:
Closing since this is working fine I believe.

closed 2017-10-18 by DonLakeFlyer