mavlink/qgroundcontrol #5742: Survey altitude not updating

JonReacher, 2017-10-13:
Haven't seen this addressed yet, when creating a new survey the survey altitude field is populated with the overall mission altitude number, however the survey grid itself is showing up using the last used survey altitude. This is using 3.2.4 on mac and android.

 So if I plan a survey at 400' agl and then close and reopen the program next time to plan a new survey, the survey will have 150' agl filled in the altitude box (the overall mission altitude set) but the survey will have the grid pattern, photo count, etc. of 400'. If I type 400 into the survey altitude field and hit enter, nothing changes. If I type any other value into the survey altitude it will change the survey to reflect the new altitude appropriately.

 One concern I have is which would be actually used, the visual grid or the listed altitude, if it didn't get manually corrected each time?

DonLakeFlyer, 2017-10-17:
I can't seem to reproduce this. Here is what I tried:
1) Boot QGC, goto Plan
2) Mission Start has Waypoint alt set to 50 meters
3) Add a survey, shows up as Manual Grid, survey altitude is set to 50 meters
4) Change the survey alt to 400
5) Shutdown QGC
6) Go back to step 1. In step 3 survey is 50 meter alt

DonLakeFlyer, 2017-12-30:
I'll reopen if you can specify more detailed steps to reproduce

closed 2017-12-30 by DonLakeFlyer