mavlink/qgroundcontrol #5746: Mission start dialog causes temporary switch to guided mode (AP 3.8.2 VTOL)

nick2204, 2017-10-16:
#### Issue details
Triggering an auto mission start using the start mission dialog/context action causes the vehicle to be set temporarily to Guided mode before it enters Auto.
This can have undesirable effects depending on what the last guided commands were.

Triggering the auto mission by setting mode to Auto and then Arming, seems to work correctly.

#### Version
Arduplane 3.8.2-dev (running in SITL)
Q_enable =1 (quadplane)
Q_guided_mode = 0 (FW loiter)
Q_RTL_mode = 1 (land in VTOL once RTL reached)

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x ] Plane
[  ] Rover
[  ] Submarine

#### Airframe type
Quadplane

#### Hardware type
PH2.1
SITL
#### Logs

fredrikfalkman, 2017-10-21:
I can confirm this behavior on a regular plan running 3.8.1 and QGC Develop add505fa7 (2017-09-17)

DonLakeFlyer, 2017-10-21:
This is currently the way it is coded to work. Looking into doing it a different way.

DonLakeFlyer, 2017-10-24:
> This can have undesirable effects depending on what the last guided commands were.

Can you explain? How could there be other guided command in play if you are on the ground?

nick2204, 2017-11-01:
It was not entirely clear what happened, @tridge may be able to offer a better explanation, but the switch happened between takeoff and getting to transition height (barely having left the ground) which led to some slight sideways navigation in hover with landing gear still partially in contact with the ground. 
long story short, quad-planes don't like this...

@tridge has now modified AP to prevent this from happening in the latest beta build 3.8.3-beta2 but it would be nice to not have to catch it only in the autopilot.

DonLakeFlyer, 2018-01-15:
I'll take a look at changing the way ArduPilot does a mission start in 3.4.

DonLakeFlyer, 2020-11-20:
Going to need more exact details of what the problem is. The sequence of commands is:
* Change to Guided flight mode in order to Arm
* Arm
* Switch to Auto

Don't see why that isn't a valid sequence.

DonLakeFlyer, 2020-11-20:
Closing unless more details as to why this is wrong are provided

closed 2020-11-20 by DonLakeFlyer