mavlink/qgroundcontrol #5747: Windows 10 GUI poor click accuracy

nick2204, 2017-10-16:
#### Issue details
The accuracy of clicking map and menu items in the main window is very poor in the vertical direction.

It may be coincidence, but it seems that it is out by exactly the height of the windows frame title bar, as though the position the mouse pointer is being measured relative to the top of the window, and the content is being reference to the top of the frame (bottom of the title bar).

This gives the effect that you have to click below the item by about the height of the title bar.

Popup menus do not exhibit this behaviour, with normal click accuracy.

#### Hardware
Windows 10 pro (1703) 64Bit
Intel HD graphics 530
NVIDA GeForce GTX 960M

DonLakeFlyer, 2017-10-16:
The menu bar isn't showing. Can you try using on of the other software rendered QGC startup shotcuts to see if it happens there as well.

nick2204, 2017-10-17:
Yes, as the issue suggested, if I bring the mouse cursor below the software rendered buttons by about the height of the menu bar, I can get them to come up. It is as though it expects the menu there but Windows doesn't render it, leaving an offset for the mouse click

NaterGator, 2017-10-17:
@nick2204  what Don is asking is that you try starting QGC with one of the shortcuts that helps to maximize GPU compatibility. There is both GPU compatibility mode and GPU Safe Mode (full software rendering, will use a bunch of battery). You should be able to find them in the start menu.

If one of those work then it is likely Qt is incorrectly detecting the active GPU (which wouldn't be too surprising with a multi-GPU setup which switches dynamically for power savings).

nick2204, 2017-10-17:
@natergator , ok will do. Never noticed them before, always just type the first couple of letters to start it from the start menu.

nick2204, 2017-10-18:
yes, in GPU compatibility mode it is resolved. Should i close this issue, or is this something that is likely to be worked on?

DonLakeFlyer, 2017-10-18:
Leave it open. It is more than likely some sort of Qt bug in 5.9. Not sure if there is a workaround. Maybe be video driver specific.

HTRamsey, 2024-06-03:
Seems long out of date. Please reopen if this is still relevant

closed 2024-06-03 by HTRamsey