mavlink/qgroundcontrol #5748: *** Feature request *** add "change loiter radius" action for Guided mode

nick2204, 2017-10-16:
#### Issue Details

In a similar way to how Altitude is set in guided mode, it would be very beneficial to FW flight profiles to have an easy method to set the loiter radius.

DonLakeFlyer, 2017-10-17:
Wondering why you would need to change this on a per-flight basis?

nick2204, 2017-10-17:
When flying fixed wing, the two main loiter profiles would be minimum safe radius for holding a specific position, ie in guided flight. But then there are cases where you might want to hold a wide loiter around a point of interest and keep a camera pointed toward the center. Situational awareness type of application.

Antiheavy, 2017-10-17:
> But then there are cases where you might want to hold a wide loiter around a point of interest and keep a camera pointed toward the center.

You could accomplish this with a Loiter waypoint in a mission plan.  Switching to Plan view, adding/inserting the waypoint(s), changing the type to Loiter, then changing the altitude and radius, then uploading, then switching views, then changing flight modes, then making the waypoint active -- all of that takes a lot of clicks and is pretty clunky in QGC, but technically you can get the job done.

Now that I type all that out, sheesh....   Yeah, a way easier way would be to just use a GoTo loiter point with easy way to adjust radius and altitude.  That would be WAY more useful for live video type fixed wing applications.

For what it is worth, I'm also a fan of having the change altitude slider shown immediately when in a Pause/Hold mode. (and now maybe a new change radius slider or even drag-able circle graphical element around the GoTo point on the map).

nick2204, 2017-10-17:
Ha ha, yeah it's only as you use guided mode more, you see how useful these things are. Auto missions work great in qgc, but they're not always suitable for more adhoc tasks.

closed 2026-10-04 by github-actions[bot]