mavlink/qgroundcontrol #5752: Import polygone

AluBis, 2017-10-18:
Hi,
I am new on this so not sure if this is the right place for my question.
But: I am looking for to import a polygon (.shp, .kml, .poly,...) to create a survey plan with a zone corresponding to precise coordinate. 
Mission Planner do this and it is really necessary for my activity: I realize photogrammetry for archeology, so there is no "mark", no building in satellite view to help me and I can't fly on a approximative zone.
There is any way to do it ?

DonLakeFlyer, 2017-10-18:
Not yet supported. But coming in the 3.3 release.

AluBis, 2017-10-20:
OK, it's a good news thanks ! But I don't understand why it is so difficult to find this features in almost all software.

Antiheavy, 2017-10-21:
+1 for this feature request.

kikislater, 2017-10-23:
+1 also

RobDeBagel, 2017-11-02:
+1 for this also. I don't know the codebase or your plans that well but I'd suggest GDAL/OGR for import/export duties as it abstracts away this problem for pretty much every raster/vector GIS data format under the sun.

DonLakeFlyer, 2017-11-02:
Also https://github.com/google/libkml

Antiheavy, 2017-11-02:
Shape files are of particular importance in some industries. https://en.wikipedia.org/wiki/Shapefile

kikislater, 2017-11-03:
Shapefile will die, gdal/ogr seems a good option IMO (there is some GIS users using Qgroundcontrol) and regarding this also it makes senses : gdal have a driver for WMS : 
http://discuss.px4.io/t/does-qgc-get-a-wms-client-any-time-soon/4788

DonLakeFlyer, 2017-11-11:
I looked at integrating a kml reading library. Trying to get something like that to work cross-platform turned out to be a nightmare. So instead a wrote some very simple KML polygon loading code. I tested with polygon output from Google Earth and it works. But not sure if there are any quirks in other Polygon formats I may not have taken into account. So testing with any KML polygon files you have would be appreciated.

Also I'm going to try to get away with kml only support in 3.3. May look at other formats in later releases.

DonLakeFlyer, 2018-03-23:
KML is in 3.3 release. SHP will be in 3.4

tubeme, 2018-07-06:
One quick question. I use the 3.3.2 and don't seem to find a way to import kml... @DonLakeFlyer can you give me a pointer? thanks.

DonLakeFlyer, 2018-07-06:
You need to use a daily build

tubeme, 2018-07-06:
OK thanks I saw it in the Master. It translates directly in corridor survey. Is there a way to import the KML as a simple line?

What is the plan with KMZ? This is a must for the mapping guys. We met a lot of them because we tailor our craft toward the mapping. And their first question always is KLM, KMZ at least. Some ask for .shp and .poly.

DonLakeFlyer, 2018-07-06:
> It translates directly in corridor survey. Is there a way to import the KML as a simple line?

Corridor Scan works with KML which has a line in it, not a polygon.

> What is the plan with KMZ?

None at the moment due to lack of cross platform unzip support. Users just have to unzip the KMZ themselves.

tubeme, 2018-07-07:
Yes I tried the KML import. My question was can I import the KML not as a survey but as a straight line with  waypoints. Like a regular flight. I did not find a way to make it...

But then if we extract the KML from the KMZ manually can we generate a survey polygon if we import the KML in QGC?

DonLakeFlyer, 2018-07-08:
> My question was can I import the KML not as a survey but as a straight line with waypoints.

Ah, nope not supported. Can you enter a separate issue with that as a feature request. Would be easy to do.

DonLakeFlyer, 2019-01-19:
Both KML and SHP are supported now. I don't think I"m going to support .poly

closed 2019-01-19 by DonLakeFlyer