mavlink/qgroundcontrol #5754: Compass calibration fails

AndKe, 2017-10-18:
I have previously reproduced it on two ArduCopter 3.5.x on Pixhawk 2.1
Today on another AC3.5.3, Pixhawk 2.1
No amonut of retries seem to help.
What is happening ?

[image]

AndKe, 2017-10-18:
related code in ArduPilot https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_Calibration.cpp#L268

DonLakeFlyer, 2017-10-20:
Firmware is rejecting the command for some reason.

DonLakeFlyer, 2017-10-20:
Do you have any of the compasses disabled?

AndKe, 2017-10-20:
Yes, 
I am posting all relevant params of the last UAV I had trouble with:

COMPASS_CAL_FIT,100.00000000
COMPASS_DEV_ID,466441
COMPASS_DEV_ID2,131874
COMPASS_DEV_ID3,263178
COMPASS_EXTERN2,0
COMPASS_EXTERN3,0
COMPASS_EXTERNAL,1
COMPASS_LEARN,0
COMPASS_PRIMARY,0
COMPASS_USE,1
COMPASS_USE2,0
COMPASS_USE3,0

So, basically, I am using only the external one.
Also, before calibration I see info like this (screenshot from Pixhawk "1", 2.1 have one more)
[image]

I find that somewhat confusing: Does the "use compass" actually change COMPASS_USE*  , or is it really a bitmask for which compasses to calibrate ? (or both ?)    
If it's a toggle for COMPASS_USE parameter, I would expect all three to have it, if it's which to calibrate, I would prefer to know if they are enabled at all.
In any case, a auto-reboot option in GUI would be nice, as it's a mandatory step after calibration.

DonLakeFlyer, 2017-10-20:
> Does the "use compass" actually change COMPASS_USE*

Yes it does. Can you try calibrating will all compasses turned on?

AndKe, 2017-10-20:
I just returned for the weekend and the UAV is few thousand kilometers away, I can try early next week. :)  - don't have a 2.1 + external GPS for it at home.

AndKe, 2017-10-21:
I had a 2.1 + external magentometer around anyway, just tested it.
Yes, checking all checkboxes works.  :)

DonLakeFlyer, 2017-10-21:
https://github.com/ArduPilot/ardupilot/issues/7058#issuecomment-335979336

closed 2017-10-21 by DonLakeFlyer