mavlink/qgroundcontrol #5775: Planned mission disappears when entering a guided go-to location

fredrikfalkman, 2017-10-22:
When flying a mission in auto, if I enter a go-to location (Guided), the mission disappears from the chart screen. If I go back to Auto, the plane reverts to following the planned mission, but the waypoints and route lines are still not showing.

Other than that, the new ability to use go-to locations seems to work nicely. Much appreciated!

3.8 (Development add505fa7 2+17-09-17)
RPi3/Navio2; Arduplane 3.8.1

DonLakeFlyer, 2017-10-25:
Can you try a new daily build? New one should show up in a couple hours.

fredrikfalkman, 2017-10-27:
011fe913d 2017-10-25 doesn't seem to fix this. Also, sometimes an old goto marker would persist (graphically) even after loading a new mission.

DonLakeFlyer, 2017-11-02:
If you are using Android the daily builds aren't updating right now so they are old. If no, can you try a newer daily. I think you may have just missed the fix.

fredrikfalkman, 2017-11-02:
I’m using the Mac version, but I’ll try a new download non the less. Sometime next week. thanks!

fredrikfalkman, 2017-11-08:
Today I had a chance to test again, this time with the Mac version ff3ee095c of 2017-11-04. Still the same thing. 

When I set a Go-to, the missions disappears:
[image]

When I return to auto the plane flies the mission happily but the waypoints aren't showing (but the goto still is:
[image]

DonLakeFlyer, 2018-01-04:
I think I fixed it for real this time!

closed 2018-01-04 by DonLakeFlyer

fredrikfalkman, 2018-01-04:
@DonLakeFlyer you’re awesome! Will give it a go next week. Thanks!