mavlink/qgroundcontrol #5779: PX4 RC calibration possible RC1 issue

dagar, 2017-10-23:
I'm seeing a lot of logs where even after RC calibration the RC1_MAX, RC1_MIN, etc parameters are still set to default. This could often go unnoticed in many setups.

http://discuss.px4.io/t/incorrect-throttle-behavior/4712

Needs further investigation, but so far looks like all of this happens on QGC side.

DonLakeFlyer, 2017-10-26:
I just did radio cal using both daily and stable. I did notice that someone changed the param meta data units for those values to "us" which is meaningless to QGC. Hence they show up as floating point with 3 digits of precision. These values should not have units associated with them.

DonLakeFlyer, 2017-10-26:
[image]

dagar, 2017-10-26:
It's meaningless as these are almost always set by the QGC calibration, but isn't it still the correct metadata, or at least the unit? They should be integers, but changing that will break every RC calibration.
Reverse could be an enum.

dagar, 2017-10-26:
I'll take note if I see this happen again. It could be failing at some other level (PX4 param save or mavlink transmission?).

DonLakeFlyer, 2017-10-26:
Oh, I thought they changed from integers. I could have sworn they used to be integers. If not everything is ok with that.

It could theoretically fail when QGC sends the new values to the vehicle. But you will get a QGC error if that happens since the vehicle will not ack back the values.

It seems very strange that firmware side would drop a specific channels set of values. Which would seem to point more to QGC. But I just don't see anything on the QGC side. Could possibly be related to a different stick mapping than I am testing with?

dagar, 2017-10-27:
I'm not sure, but it's currently working fine for me. We'll just have to keep an eye out for it.

closed 2017-10-27 by dagar