mavlink/qgroundcontrol #5783: Survey: Deselect not disabling images in turnaround

JonReacher, 2017-10-24:
I have noticed, using 3.2.4, that leaving "take images in turnarounds" not checked does not keep the images from being taken during the turns in surveys. I have had this happen on all of my surveys, along with some inconsistent trigger distances leaving large gaps in coverage.

flight log: https://logs.px4.io/plot_app?log=ed0c74f0-6b5b-4c4b-8a60-e6935887978f

I re-downloaded the mission from the autopilot after seeing this to make sure that I didn't accidentally have it checked. I see that upon entering the turn it changes the image distance to 0. Would it make sense to change to 'stop taking photos'?

DonLakeFlyer, 2017-10-24:
> I see that upon entering the turn it changes the image distance to 0. Would it make sense to change to 'stop taking photos'?

Those are exactly the same thing. I also assume you meant "exiting" not "entering"?

What firmware are you using? Can you attach the .plan file which is not working.

JonReacher, 2017-10-24:
Here's the plan file. PX4 firmware ver 1.6.5
[Sand.plan.zip](https://github.com/mavlink/qgroundcontrol/files/1412358/Sand.plan.zip)

 I wasn't sure if they equaled the same thing or not, so it sounds like that's fine. I was meaning exiting the survey polygon/ entering the turnaround.

DonLakeFlyer, 2017-10-25:
Plan file persistance for images in turnaround is broken. If you change it from the ui and then upload mission will be correct. But if you load from file it will be wrong.

JonReacher, 2017-10-27:
@DonLakeFlyer Any thought on the trigger pattern? If it was intending to shoot through the turnarounds I wonder what caused the gaps.

DonLakeFlyer, 2017-10-27:
Gaps in image capture? Could be trying to take images faster than the camera can do. Other than QGC turning on/off the camera at the appropriate spots and setting the trigger distance after that the firmware is in charge of running the mission.

JonReacher, 2017-10-27:
The camera is controlled via IR trigger, so there's no way for QGC or PX4 to know if the camera is at it's buffer limit, but the log shows big gaps in coverage of the trigger events themselves (let alone the actual images taken).

DonLakeFlyer, 2017-10-28:
QGC image turnaround persistance is fixed now in Daily and Stable. Closing since from the QGC side things are working correctly. You'll need to talk to the firmware golks about any missed triggers.

closed 2017-10-28 by DonLakeFlyer