mavlink/qgroundcontrol #5785: iOS Build:  Xcode build fail with  "CFBundleShortVersionString"

khoinguyen258, 2017-10-25:
Hi all,
I am trying to build qgroundcontrol for IOS. Following the tutorial on this page. I create a xcode file by using qmake (from qtcreator ) . However when I build the code with Xcode , I got the following error:

> Set: Entry, ":CFBundleShortVersionString", Does Not Exist
Although I try many way to change the CFBundle value in IOS-info .plist.
I do not know whether this is a issue or only error on my machine. 

I am working on Mac OS Sierra and IOS 11 SDK

DonLakeFlyer, 2017-10-25:
Here is the text you deleted from the Issue:
Note: GitHub Issues are for bugs or feature requests only. Do not enter Issues for questions regarding usage or problems **building** QGC yourself. 
Use the appropriate support channel instead: http://qgroundcontrol.com/#resources.

closed 2017-10-25 by DonLakeFlyer

MyOnlyOnlySunshine, 2018-09-07:
I have the same error with that one ,can anybody tell me the reason and help me to solve it ,thanks .