mavlink/qgroundcontrol #5787: Android daily builds are not updating

DonLakeFlyer, 2017-10-25:


tve, 2017-11-02:
Looks like the problem is:
```
googleapiclient.errors.HttpError: <HttpError 403 when requesting https://www.googleapis.com/upload/androidpublisher/v2/applications/org.mavlink.qgroundcontrolbeta/edits/17445856360346861245/apks?uploadType=media&alt=json returned "APK is marked as debuggable.">
```
but you probably already saw that...

tve, 2017-11-02:
I'm trying to download the latest Android daily from the S3 bucket. Is there a reason listing the S3 bucket files is not allowed? That makes it hard to locate anything...

DonLakeFlyer, 2017-11-02:
@dagar Not sure where you are at on getting dailies updating again. Should we just post the S3 link in the docs?

dagar, 2017-11-06:
We can list the S3 link if people want. I still want to get the Play store fixed. I was working on pushing the pieces into a docker container to remove some of the complexity.

dagar, 2017-11-17:
Did you see my message in slack? This commit broke the google play deploy. https://github.com/mavlink/qgroundcontrol/commit/cc1fa95942fdc31a8445275b75ec88b19dba0b88

DonLakeFlyer, 2017-11-17:
No I missed that. If I remember I was having problems debugging on one of my devices. I think you can revert that without problem.

closed 2017-11-22 by dagar

DonLakeFlyer, 2017-12-09:
Awesome, thanks!

PR #5855 (2017-11-22): Revert "Add debuggable to manifest"
This reverts commit cc1fa95942fdc31a8445275b75ec88b19dba0b88.

Fixes #5787
commit 75962571 (2017-11-22): Revert "Add debuggable to manifest" (#5855)

This reverts commit cc1fa95942fdc31a8445275b75ec88b19dba0b88.

Fixes #5787