mavlink/qgroundcontrol #5791: COM Port is not re-establishing after Reboot command through QGC

sujiv0, 2017-10-26:
I'm using QGC V3.2.4 in Windows 7. After giving reboot command from Settings --> Parameters --> Tools --> Reboot Vehicle, Pixhawk 2.1 (with cube) is not connecting to QGC and showing "Link Error". I'm using latest PX4 firmware.

[image]

@DonLakeFlyer Please have a look.

DonLakeFlyer, 2017-10-28:
Ah, windows ... sigh

sujiv0, 2017-10-30:
I sit okay now??? @DonLakeFlyer

DonLakeFlyer, 2017-10-30:
I would assume not.

DonLakeFlyer, 2017-12-29:
@sujiv0 Can you try this again using latest daily build? Lorenz just put through a fix that might take are of this.

DonLakeFlyer, 2018-01-13:
Just tested with latest daily and working fine

closed 2018-01-13 by DonLakeFlyer