mavlink/qgroundcontrol #5792: PX4 px4fmu-v2/px4fmu-v3 flashing issues

dagar, 2017-10-26:
px4fmu-v2 boards that report 2MB usable flash should be using px4fmu-v3. This was working previously.

dagar, 2017-10-26:
[image]

DonLakeFlyer, 2017-10-28:
@LorenzMeier Is there some way for me to get one of these boards for testing? I don't have anything that is v3.

LorenzMeier, 2017-10-29:
@dagar Pixhawk 2.1 is getting correctly detected. What board are you using?

LorenzMeier, 2017-10-29:
@dagar could you briefly debug?

https://github.com/mavlink/qgroundcontrol/blob/master/src/VehicleSetup/Bootloader.cc#L586-L591

You can get output like this in Qt Creator:

```qDebug() << variable;```

Dnrodriguez374, 2017-11-03:
I am having a similar issue. I cannot re-flash my pixhawks successfully with QGC stable or daily build

Here is where it gets stuck at every time. 
[image]

DonLakeFlyer, 2017-11-04:
@Dnrodriguez374 What type/brand of boards are you flashing?

Dnrodriguez374, 2017-11-04:
@DonLakeFlyer the pixhawks are from MRobotics.

dagar, 2017-11-06:
This is working correctly with my pixhawk 2 cube.

[image]

dagar, 2017-11-06:
The original screenshot was from the Dronecode test team using a Pixhawk mini.

bkueng, 2017-11-22:
I can reproduce this on a Pixhawk Mini: It's recognized as V2 HW. The following instrumentation:
```
    qWarning() << _boardID << ", "<<_bootloaderVersion << ", " << _boardFlashSize;
```
outputs:
```
9 ,  4 ,  2080768
```

So the bootloader version is 4, but we need at least version 5 to reliably detect V3.

dagar, 2017-11-22:
Thanks @bkueng 
That means we'll need to fix it on the firmware side by squeezing proper pixhawk mini support into the 1MB px4fmu-v2 binary. https://github.com/PX4/Firmware/issues/8342

closed 2017-11-22 by dagar