mavlink/qgroundcontrol #5793: Request: Hotkey or button to "create WP at current position"

AndKe, 2017-10-26:
When in planning mode, Vehicle is shown on map.
We often fly over complex areas to collect outer boundaries of late-to-be surveyed area.  
(looking nadir to determine the affected area before planning survey)

It would be very neat, if we had a key/button to create a waypoint at UAV's current lat/lon/alt.  
(in QGC, not UAV)

fingadar, 2017-10-27:
Good idea. We often have the problem that "normal" users have problems to set the polygon points for survey grid using the map.
Creating the polygon points for a survey grid with the actual position of the tablet would be perfect for them. So they could simply walk around the borders of their agricultural field, set the points simply by pushing a button on the pad and that would be perfect.
DJI showns this method in one of their promotion videos for their Argas copter.

DonLakeFlyer, 2019-01-19:
There isn't a hotkey but you can select Edit Location from the hambuger which allows you to move an item to the vehicle location.

Williangalvani, 2019-06-04:
This seems to the same as #4275

AndKe, 2019-06-04:
Yes, those are basically the same.
Should be labeled as for Release 3.9 ,that would be more believable, right along with proper flight monitoring features requested years ago :)

DonLakeFlyer, 2019-06-05:
> proper flight monitoring features

Don't remember what that is. Is there an issue somewhere?