mavlink/qgroundcontrol #5794: "Go to Location" should contain altitude.

AndKe, 2017-10-26:
Screenshot says all:
"Goto Location" should display, and offer the slider to the right to change altitude.  - those two actions are very often used together.

[image]

DonLakeFlyer, 2017-10-26:
Yeah, makes a ton of sense.

DonLakeFlyer, 2017-10-28:
Same thing for takeoff

closed 2018-06-20 by DonLakeFlyer

AndKe, 2018-06-20:
Unable to verify, - the "takeoff altitude" is still missing
Guided is impossible to use (current master)  - any left/right click pan map in a .. inverse way.
https://youtu.be/RPfuwqN4blk

DonLakeFlyer, 2018-06-20:
This should not have been closed. The other scrolling thing is quite bizarre!

DonLakeFlyer, 2018-06-20:
Can you enter a separate issue for the click thing?

AndKe, 2018-06-21:
Done: https://github.com/mavlink/qgroundcontrol/issues/6625   :)