mavlink/qgroundcontrol #5795: "Change Altitude" limits, linearity, and altitude type.

AndKe, 2017-10-26:
In the screenshot below:
1: moving slider up, does nothing, top and center of the  is ~122 meters (400ft)

This is not usable for a lot of missions. even for simple missions in G airspace, below 122m AGL, we may start at sea level, and while being <122m AGL, we are flying at >300 relative altitude.  , or the other way around, launching from 500mAMSL, flying down into a valley, the survey may be at a relative altitude of -400m

to sum up, the slider would be useful if it always presented current altitude as center, and top/bottom as +200 / -200 meters, maybe a logarithmic slider would be better, controlling speed of change, rather than altitude.

2: an numeric text entry would be nice as well.
3: please offer to toggle REL of ABS altitude, which one we would use depends on the mission.  Another option would be to just display both.

[image]

Antiheavy, 2018-01-18:
@AndKe I agree the current implementation of the altitude slider isn't quite right.  It feels like a hybrid of a relative altitude slider and an absolute altitude slider, but works poorly no matter which way you are thinking about it.   Part of the problem is trying to address the different needs of multicopters and fixed wings.

I like the idea for a toggle between absolute and relative slider styles.

For what is it worth this is an idea I had a while back for a more detailed absolute altitude slider:

[image]

AndKe, 2018-01-18:
looks/sounds good :)

DonLakeFlyer, 2018-06-20:
I think this is better now. But may still need work.

AndKe, 2018-06-20:
I just built current master, and nothing seems to be changed ?

DonLakeFlyer, 2018-06-20:
Thought there was more. Only logarithmic is there.

Antiheavy, 2018-06-22:
> Only logarithmic is there.

The currently implemented logarithmic scale doesn't work well for large altitude changes.  In our fixed wing use case we'd like to be able to quickly adjust 50m-100m up or down (usually down) to avoid overflying aircraft.

DonLakeFlyer, 2020-11-08:
Min/Max altitude are configurable in settings. The non-linear slider is going to stay as is.

closed 2020-11-08 by DonLakeFlyer